EttusResearch/uhd #142: PCIe does not work if send buffer size is not multiple of 1020

mdmikh, 2017-11-03:
It was noticed that transmitting of samples through PCIe interface is seriously broken if buffer size is not multiple of 1020 samples and receiving is performed simultaneously. Attached test program [test3.zip](https://github.com/EttusResearch/uhd/files/1441768/test3.zip) demonstrates that.

The test tries to receive and send data using buffer of 15360 samples (15 full packets by 4096 bytes + one partial packet). A lot of 'L' characters is printed by UHD indicating that the streaming is broken. If buffer size 15300 is used there are no any errors.

$ sudo ./test3
linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.002.000-5-g4bbf0c20

Found type=x300,resource=RIO0,product=X310,fpga=HG
...
Max samples per rx packet = 1020
Max samples per tx packet = 1020
Streaming...
ULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLL
...

There are no errors also with  10G Ethernet interface.

In my opinion the issue is serious because PCIe interface is not usable due to that bug.

michael-west, 2017-11-15:
Thank you for posting this issue and the thorough explanation and example.  We will look into it and try to resolve it as soon as we can.  In the meantime, please pad buffers to multiples of 1020 samples when using PCIe.

michael-west, 2018-03-27:
Moved to internal issue tracker for triage and troubleshooting.

closed 2018-03-27 by michael-west

michael-west, 2018-08-24:
Tested on v3.13.0.2.  The issue seems to be resolved.  Thanks again for posting, @mdmikh !