EttusResearch/uhd #144: uhd::transport::sph::send_packet_streamer_send() lead to Segmentation fault

liuzhikui, 2017-11-28:
When I run USRP B210,after some time,**uhd::transport::sph::send_packet_streamer_send()** triggered **Segmentation fault**
Here is the backtrace:
[image]

michael-west, 2018-02-10:
A segmentation fault of this type can be (and usually is) caused by invalid parameters to the uhd_tx_streamer_send() call.  There can be other causes, but we are unable to reproduce the issue without more detail.

Closing this issue for now.  If you want to re-open it, please provide detail on how to reproduce the error.

closed 2018-02-10 by michael-west

michael-west, 2018-04-26:
Root cause identified in #171.