EttusResearch/uhd #147: Build broken with boost 1.66.0

fxcoudert, 2017-12-19:
```
/tmp/uhd-20171219-86213-y0p3fu/uhd-release_003_010_002_000/host/lib/transport/udp_zero_copy.cpp:193:29: error: no member named 'native' in 'boost::asio::basic_datagram_socket<boost::asio::ip::udp>'
        _sock_fd = _socket->native();
                   ~~~~~~~  ^
```

`basic_datagram_socket::native` has been [deprecated](http://www.boost.org/doc/libs/1_47_0/doc/html/boost_asio/reference/basic_datagram_socket/native.html) since Boost 1.47.0, released July 11th, 2011. It should be [`native_handle`](http://www.boost.org/doc/libs/1_66_0/doc/html/boost_asio/reference/basic_datagram_socket/native_handle.html) instead.

michaelld, 2017-12-19:
See also this thread on USRP users: http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2017-December/027268.html

ilovezfs, 2017-12-19:
See https://github.com/EttusResearch/uhd/pull/148

mbr0wn, 2017-12-19:
@fxcoudert @ilovezfs Thanks for the detailed description and the suggested fix. Goes onto master immediately, and maint and LTS will follow.

closed 2017-12-19 by mbr0wn

michaelld, 2017-12-19:
@mbr0wn also need to remove boost versions 1.46 and 1.46.0 from the top-level CMakeLists.txt file, as "native_handle" was introduced in 1.47.0 (as mentioned above).

mbr0wn, 2017-12-19:
Ah, I didn't realize the implications. We can't do that, of course, we have a "contract" that we don't remove Boost versions mid-release. That means we can't actually fix this the same way as we did on master. We'll need to `#ifdef` around those calls.

Bringing to attention of @michael-west.

mbr0wn, 2017-12-19:
```
#if BOOST_VERSION < 104700                                                                                                                                      
    // use native()
#else
    // use native_handle()
#endif

We've had to resolve to this in the past (https://github.com/EttusResearch/uhddev/blob/8138e3947f4ba935f52329644c44b58da34c45fe/host/lib/transport/nirio/rpc/rpc_client.cpp#L59-L73).

mbr0wn, 2017-12-19:
Requires
```cpp
#include <boost/version.hpp>
```

mbr0wn, 2017-12-20:
maint is actually on 1.53 minimum Boost, so we can just cherry-pick. LTS allows for Boost 1.46.

michael-west, 2017-12-20:
On maint.  LTS required additional code and will be merged soon.