EttusResearch/uhd #149: How to add new device to uhd

KcMeterCEC, 2017-12-27:
Hi guys,
I have made my new SDR board which is controlled by embedded linux.
> 1. The board is communicate with PC through USB3.0(FTDI), the USB3.0 driver is very simple.

I found the uhd library is an **open source and free** library, so i want to add my SDR board driver to the library.
I have read **UHD Development Manual** which didn't indicate how to port a new device to library.
And I have read the source code, I found the code about  device is located at **uhd/host/lib/usrp**.

But it still make me confused about the **CHDR protocol** and device API.

How can i get a cleaner and simple step to add my device?

mbr0wn, 2018-01-03:
Either maintain your own branch of UHD, or use the plugin API. Please understand that we don't have resources to support other people's hardware.

closed 2018-01-03 by mbr0wn

FFY00, 2018-01-04:
You can write your software for SoapySDR which has a UHD plugin making it compatible with the UHD driver.