EttusResearch/uhd #150: tx_samples_from_file: bandwidth typo

pyBlob, 2018-01-13:
the tx_samples_from_file displays the filter bandwidth in Hz instead of MHz:
https://github.com/EttusResearch/uhd/blob/maint/host/examples/tx_samples_from_file.cpp#L153

the other files like this one seem to convert the value to MHz before printing:
https://github.com/EttusResearch/uhd/blob/maint/host/examples/txrx_loopback_to_file.cpp#L377

`# tx_samples_from_file --file sband_tx16 --rate 25000000 --freq 2275900000 --ant TX/RX --bw 1000000 --gain 24
linux; GNU C++ version 7.2.1 20171025; Boost_106200; UHD_003.010.002.000-3-g122bfae1

UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam

Creating the usrp device with: ...
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes

UHD Warning:
    The send buffer could not be resized sufficiently.
    Target sock buff size: 1048576 bytes.
    Actual sock buff size: 288000 bytes.
    See the transport application notes on buffer resizing.
    Please run: sudo sysctl -w net.core.wmem_max=1048576

UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam
Using Device: Single USRP:
  Device: USRP2 / N-Series Device
  Mboard 0: N210r4
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: SBXv3 RX
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: SBXv3 TX

Setting TX Rate: 25.000000 Msps...
Actual TX Rate: 25.000000 Msps...

Setting TX Freq: 2275.900000 MHz...
Actual TX Freq: 2275.900000 MHz...

Setting TX Gain: 24.000000 dB...
Actual TX Gain: 24.000000 dB...

**Setting TX Bandwidth: 1000000.000000 MHz...
Actual TX Bandwidth: 1000000.000000 MHz...**

Checking TX: LO: locked ...

Done!
`

mbr0wn, 2018-01-15:
@pyBlob thanks a lot for pointing this out! We have a local fix here and it should go public pretty soon.

mbr0wn, 2018-02-08:
eaed9d5ac511a08a123a28d1c0fc48789ca1480b

closed 2018-02-08 by mbr0wn