EttusResearch/uhd #153: cmake error

robellebor, 2018-02-08:


robellebor, 2018-02-08:
robel@robel-OptiPlex-760:~/workarea-uhd/uhd/host/build$  cmake -DPYTHON_LIBRARIES=/Library/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib .
-- 
-- Configuring the python interpreter...
-- Python interpreter: /usr/bin/python3
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Working off of feature or development branch. Updating version number.
-- Using UHD Images Directory: OFF
-- Build type not specified: defaulting to release.
-- 
-- Configuring Boost C++ Libraries...
-- Could NOT find Boost
-- Boost include directories: /usr/include
-- Boost library directories: /usr/lib/x86_64-linux-gnu
-- Boost libraries: 
-- 
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
-- 
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - found
-- 
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - found
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 0
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = TRUE
CMake Error at cmake/Modules/UHDComponent.cmake:51 (MESSAGE):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:351 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/robel/workarea-uhd/uhd/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/robel/workarea-uhd/uhd/host/build/CMakeFiles/CMakeError.log".

i use ubuntu 17.1

robellebor, 2018-02-08:
thanks i fixed by changing 

  "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" OFF ON)
to this 

  "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" ON ON)

closed 2018-02-08 by robellebor

dkozel, 2018-02-08:
@robellebor That looks like you do not have all the dependencies installed. Here is a guide which contains all the dependencies listed. The Ubuntu 17.04 list will almost certainly work for 17.10.

https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux#Update_and_Install_dependencies

I hope this works for you. If you have any other questions the mailing list is a great resource and has a good community.

audiohacked, 2018-02-11:
@robellebor @dkozel The problem is Ubuntu has a different naming scheme for the libboost_python library so CMake is unable to find the proper libraries. Normal naming scheme is "libboost_python2.7" and "libboost_python3.5" but Ubuntu names the libaries "libboost_python-py27" and "libboost_python-py35"

noc0lour, 2018-02-11:
@audiohacked If that is the case this is most likely a `cmake` bug in Ubuntu since the functionality in the distribution shipped `FindBoost` `cmake` is used. If Ubuntu changes naming for Boost libraries it should account for it in the `cmake` Module.

mbr0wn, 2018-02-12:
Hm yeah, this is hard to fix from our end.

audiohacked, 2018-02-12:
@mbr0wn One way to "fix" it is to include a patched FindBoost.cmake module in uhd.

rjzak, 2019-05-08:
I had this same error, and had to edit the cmake file to ON ON in the Win32 If/Else block. It seems stupid that the else block disables the Python API, despite the required libraries being found.

mbr0wn, 2019-05-09:
Boost.Python was removed from UHD (and replaced with PyBind11).

rjzak, 2019-05-09:
In which version? I'm pretty sure that's not documented. I was going over the compilation instructions.

mbr0wn, 2019-05-09:
You're right, it's on master. Anyhow, I'm not sure what your exact problem is (given the history of this issue), can you please open a new one?

awilliamson, 2019-07-13:
This definitely needs to be fixed in documentation, or more easily accessible. This has bitten me twice now, from two separate installations.

joeldelosangeles, 2021-09-22:
> thanks i fixed by changing
> 
> "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" OFF ON)
> to this
> 
> "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" ON ON)

Which file should be edited?  Thanks

joeldelosangeles, 2021-09-22:
> > "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" OFF ON
> 
> Can please tell me how and where did you change it?

 vim /usr/local/src/uhd/host/CMakeLists.txt

hanghoo, 2023-06-21:
> thanks i fixed by changing
> 
> "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" OFF ON) to this
> 
> "Boost_FOUND;HAVE_PYTHON_PLAT_MIN_VERSION;HAVE_PYTHON_MODULE_MAKO" ON ON)

Thank you very much for sharing the solutions. Could you explain further why we have to change the OFF to ON?

kraj, 2023-07-08:
I think you are missing a dependency `python-mako` once its installed on the system this error should go away

mbr0wn, 2023-07-10:
Yeah you really don't have to modify CMakeLists.txt files if you're on a standard Ubuntu or similar

K0zero, 2023-08-29:
> kraj

this is worked to me  thanks a lot

RRoHES, 2024-02-18:
I did pip install mako to resolve this issue which download the missing mako module