EttusResearch/uhd #156: set_master_rate() doesn't work on x300

aholtzma, 2018-02-16:
It appears that set_master_rate doesn't work correctly on x300. The rate gets 'set' and you when you get it, it is correct. However the code in x300_clk_cntrl that sets up the PLL with the master clock frequency doesn't get called. It seems to be only called once, in the contractor for the object.

michael-west, 2018-02-16:
@aholtzma This is by design.  X300 does not support dynamically changing the master clock rate.  To set it, just add the device argument "master_clock_rate=<rate>" when creating the multi_usrp object.  See the manual [here](http://files.ettus.com/manual/page_configuration.html#config_devaddr).

An issue to improve the API call for X300 has been added to our internal issue tracker.  Thank you for reporting it!

aholtzma, 2018-02-20:
Is there a document with all of the unsupported API calls for x300?

This one is particularly evil in that it not only doesn't return an error, it changes internal state so that when you call get_master_clock_rate() it returns what you set (and not the actual rate). It might even be in a half-way state, where some parts think it is using new rate, and half the original rate.

mbr0wn, 2018-02-20:
@aholtzma All API calls should either coerce or throw exceptions. In the case of the X300 we need to make sure that either happens.

Either behaviour (coercion or throwing) needs to be documented in the `multi_usrp` API docs, otherwise, I'd consider that a bug. Thanks for bringing this up!

aholtzma, 2018-02-20:
Ok glad to know this will get cleaned up.

What is the restriction on dynamic clock rate changes on x300? It looks to me like you should be able reprogram the PLLs and everything else should be able to carry on w/o noticing.

mbr0wn, 2018-03-26:
@aholtzma We'll have a fix out soon. Note that the contract of `multi_usrp::set_master_clock_rate()` states that it will coerce, and not throw, so you need to run `get_master_clock_rate()` afterwards if you want to be sure.