EttusResearch/uhd #157: ETTUS: GR-UHD detected ABI compatibility mismatch with UHD library (ABI: 3.10.2,  UHD library reports ABI: 3.10.3)

kgitit, 2018-02-23:
I have a problem with gnuradio on the Linux mint

Issue :
RuntimeError:
GR-UHD detected ABI compatibility mismatch with UHD library.
GR-UHD was build against ABI: 3.10.2,
but UHD library reports ABI: 3.10.3
Suggestion: install an ABI compatible version of UHD,
or rebuild GR-UHD component against this ABI version.

I have reinstalled and removed all.
I have try 10 times at install Gnuradio and all components.

Below the steps used to install gnuradio:

    Install PyBOMBS:
    $ [sudo] pip install PyBOMBS

    Add PyBOMBS recipes:
    $ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
    $ pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git

    Configure an installation prefix:

    $ pybombs prefix init ~/prefix/default/

-Install GNU Radio:

$ pybombs install gnuradio

-Run GNU Radio Companion from your new prefix:

$ source ~/prefix/default/setup_env.sh
$ gnuradio-companion

    Source Build automatic
    sudo apt-get install gnuradio

Please let me know if anyone has found the solution

Thanks :) :) :)

dkozel, 2018-02-23:
Hi @kgitit 

We try to keep GitHub issues for bug reports or tracking feature requests rather than doing support. Could you email the usrp-users or discuss-gnuradio mailing lists for questions like this?
http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com
https://lists.gnu.org/mailman/listinfo/discuss-gnuradio

As the error message is saying the problem is that the version of UHD which is installed is newer than the one present when GNU Radio was built. The easiest way to resolve this problem is to rebuild GNU Radio without doing anything to UHD.

```
pybombs -p ~/prefix/default/ rebuild gnuradio
```

You also included `sudo apt-get install gnuradio` in your message. Using both PyBombs and apt at the same time will probably cause conflicts despite PyBombs trying to avoid it. If you have installed GNU Radio or UHD yourself using apt it is probably best to uninstall them before running pybombs again.

If this doesn't work for you could you please email one of the lists or support@ettus.com.

Thanks!

closed 2018-02-23 by dkozel

kgitit, 2018-02-23:
ok 

Thanks :)