EttusResearch/uhd #159: Problems with Ubuntu PPA

csete, 2018-03-06:
Hi guys,

I don't know whether you are aware of this, but every time UHD is updated in the Ettus PPA, it replaces the previous version and breaks all binary packages that were built using the previous version of the library. Normally, it should be possible to keep the previous version(s) installed until the packages depending on it are rebuilt, where after the old libraries are marked "auto removable". At least this is how it works with the GNU Radio packages in Debian and Ubuntu.

As far as I can tell this happens because the installed package is called libuhd003 and the package manager can not have two versions installed at the same time, since they would both be called libuhd003. Debian Testing has added the minor and micro version numbers to the package name: libuhd003.010.003 and this would allow keeping the old version since the new package would be called libuhd003.011.000.

Alex

dkozel, 2018-03-07:
Hello Alex,

Thanks for the feedback. UHD uses `Major.API.ABI.patch` as its version naming scheme, the same as GNU Radio is adopting with the upcoming 3.8.0.0 release, so Maitland's method in Debian would be sufficient protection against the problem you've encountered. Moving to the semantic versioning scheme in 3.10.0.0 was intentionally designed to let us and maintainers do intelligent things based on the release version numbers.

We're in the middle of some packaging discussions right now and I'll make sure the right people take a look at this.

csete, 2018-03-07:
Sounds good, thanks.

mbr0wn, 2018-03-13:
What's a good path forward? Launchpad overwrites the packages, is there a setting to avoid that? Maitland's method for Debian is different, because he controls what goes into which release. I'm not sure how to both publish recent UHD packages through Launchpad and keep ABI intact.

guruofquality, 2018-03-13:
Come up with a ABI number that changes whenever the binary compatibility actually changes and a version release number like you have. They can be related, but typically its not the same thing, since the version can have a patch number incremented, without changing the ABI. I have seen project using a single incrementing digit. I'm partial to major.minor since I dont change the API between patch releases.

I wrote some of this up before, so to share: https://wiki.myriadrf.org/ReleaseGuide#Version_numbers
I hope that helps, pay close attention to those two symlinks with the three so files, its there for a reason.

From **cmake**'s perspective you just do this:
```
set_target_properties(uhd PROPERTIES SOVERSION ${ABI_VERSION})
set_target_properties(uhd PROPERTIES VERSION ${RELEASE_VERSION})
```

For **debian** packaging, the control file's package entry for the library and the install file should always change to match the exact ABI of the library. When a dependent project like gqrx is packaged in a deb, it actually automatically depends on this ABI-specifically named package (aka libuhd3.12) and not the canonically name libuhd-dev package.

And so the **magic** is that if you do those two things, any release patch update with the same ABI will just update the library used by all the dependent packages. Any new released debs with new ABI will not replace the old package. So you when install libuhd-dev, you always pull in the latest library package with the latest abi, but when you install gqrx build against an older uhd, gqrx depends on the older library and installs it without any conflict because it has a unique name with the abi in it.

Now the **caveat** to this is that while libraries may coexist happily, theres only one copy of bin utils and file system resources like images and modules. If those will cause a conflict across UHD releases, I would put them into ABI specific subdirectories too. Like /usr/share/uhd/images/3.12/blah_images_here

And on that note it would be nice to see an additional **module** search path include `$prefix/lib$suffix/uhd/modules3.12/`, so that modules can be packaged in a directory that is safe to be searched only by the relevant UHD release since modules are also compiled against a specific uhd ABI much like gqrx would be.

Also on my **wishlist**, I would love to see the cmake package files (the ones that get installed) become part of the libuhd-dev package. All that work to make it easy to build dependent projects and I still have to copy that old FindUHD.cmake around because the debs are missing it.

Also, if you are looking to publish to the PPA. I don't know how everyone does it or what even is proper. But I recommend checking the debian/ directory and just versioning it. Its really easy to bump its numbers and changelog when doing a release. You can build debs from that pretty easily using gitbuildpackage to test it.

I made myself a script to upload a package to a ppa given only a git repo with a debian/ directory if you are interested: https://github.com/pothosware/scripts/blob/master/ppa_helper

And please take a look at [SoapySDR](https://github.com/pothosware/SoapySDR/) which uses a lot of the tricks mentioned such a library numbers, debian controls files, module directories with ABI specific numbers, etc...

It took a while for me to figure this out, hopefully you found some of this useful. Perhaps we can get UHD off to a friendly library scheme so myriadrf and gqrx ppas are a bit easier to maintain moving forward. We like using your PPA, its convenient.

mbr0wn, 2018-03-13:
@guruofquality Thanks for all the pointers. We'll check them out. There are a couple things we already do, and I would like to point them out:
- Versions and ABIs are matched. The first 3 digits of our UHD release is the ABI version. We use `abi-dumper` to automatically verify we're not breaking ABI
- We already version our `debian/` subdirectory. All Lintian warnings are fixed whenever we encounter them. The .debs on the PPA are properly versioned. The issue is, we blow away the previous version (and that's something I'm not so sure about)

You're right about UHDConfig.cmake not being packaged (fix is in the pipeline) and I'll check out your comments on SOVERSION.

guruofquality, 2018-03-14:
Cool, appreciate that, and glad to the see the debian stuff checked in :-)

If you take a look at the debian/control file you have `Package: libuhd003` and in the CMakeLists.txt you have `SET_TARGET_PROPERTIES(uhd PROPERTIES SOVERSION "${UHD_VERSION_MAJOR}")` So you always just have this as 003, thats why the package gets replaced.

But I see there is an ABI version number `#define UHD_VERSION_ABI_STRING "@UHD_VERSION_MAJOR@.@UHD_VERSION_API@.@UHD_VERSION_ABI@"` for version.hpp. So this is probably what you want for SOVERSION and have to find/replace in all of the debian/control and install files to match accordingly.

gravest, 2018-03-16:
Any update on this.  There are users/customers like me who are in a pickle right now because of this issue?

mbr0wn, 2018-03-27:
OK, we've spent some time learning about package management. Here's our plan:
- Put the ABI version into the SOVERSION (thanks @guruofquality for that pointer)
- Use version numbers in the package names (e.g., like Boost, although we're not quite decided on the details)

This will probably take a little while before we can implement all of this, but we'll close this issue for our tracking purposes and move it  into our internal issue tracker. Thanks all for your input!

closed 2018-03-27 by mbr0wn

mbr0wn, 2018-09-19:
@guruofquality how does this look: https://github.com/EttusResearch/uhd/pull/214