EttusResearch/uhd #161: FPGA 47 tap HB filter coefficients files missing a comma

awalls-cx18, 2018-03-08:
When researching the performance of the X310 DDC halfband filters, I noticed that all the hb47.coe files in the fpga-src tree were missing a comma on line 48, which could possibly affect the last four coefficients of the filter.  e.g.:

https://github.com/EttusResearch/fpga/blob/2fb512b116ae81cdbd4976ea8c31762a6787224c/usrp3/top/x300/coregen_dsp/hb47.coe#L48

I'm not sure if it actually does or not, as the Xilinx tools autogenerate the filters into very complicated files.

awalls-cx18, 2018-03-08:
Looking at the last 4 lines of the hbdec1.mif file, it looks like the missing comma probably didn't matter:

https://github.com/EttusResearch/fpga/blob/8808caccd257a8bbb2ab0f1ecc17d886905c02c1/usrp3/top/x300/coregen_dsp/hbdec1.mif#L44

at least for that one file.

mbr0wn, 2018-03-09:
Thanks @awalls-cx18! Yeah it seems not to matter, but we'll fix it anyway for consistency's sake. Need to run this through our CI and review process, but it's taken care of and I'll close this.

closed 2018-03-09 by mbr0wn