EttusResearch/uhd #162: Error: RuntimeError: The hardware has no gpio bank: [...]

fmeynadier, 2018-03-14:
I may have found a regression between 3.10 and 3.11, and I don’t see it reported elsewhere. I first spotted it in my own code (calling set_gpio_attr() ) but it is also visible using the provided gpio.cpp example.

Here with the latest "maint" branch code (same happens with 3.11.0) :
<pre>
$ ~/uhd/host/build/examples/gpio --bank TXA

Creating the usrp device with: ...
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.0-8-g46ebbfc5
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
Using Device: Single USRP:
  Device: USRP2 / N-Series Device
  Mboard 0: N210r4
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: BasicRX (AB)
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: LFTX (AB)

Initial GPIO values:
       Bit  10  9  8  7  6  5  4  3  2  1  0
      CTRL:  0  0  0  0  0  0  0  0  0  0  0
       DDR:  1  1  1  1  1  1  1  1  1  1  1
    ATR_0X:  0  0  0  0  0  0  0  0  0  0  0
    ATR_RX:  0  0  0  0  0  0  0  0  0  0  0
    ATR_TX:  0  0  0  0  0  0  0  0  0  0  0
    ATR_XX:  0  0  0  0  0  0  0  0  0  0  0
       OUT:  0  0  0  0  0  0  0  0  0  0  0
  READBACK:  0  0  0  0  0  0  0  0  0  0  0
Error: RuntimeError: The hardware has no gpio bank: TXA:
</pre>

And with 3.10.1 (obviously same computer and same USRP hardware):

<pre>
$ ~/uhd-release_003_010_001_001/host/build/examples/gpio --bank TXA
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.001-0-unknown

Creating the usrp device with: ...
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes
Using Device: Single USRP:
  Device: USRP2 / N-Series Device
  Mboard 0: N210r4
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: BasicRX (AB)
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: LFTX (AB)

Initial GPIO values:
       Bit  10  9  8  7  6  5  4  3  2  1  0
      CTRL:  0  0  0  0  0  0  0  0  0  0  0
       DDR:  1  1  1  1  1  1  1  1  1  1  1
    ATR_0X:  0  0  0  0  0  0  0  0  0  0  0
    ATR_RX:  0  0  0  0  0  0  0  0  0  0  0
    ATR_TX:  0  0  0  0  0  0  0  0  0  0  0
    ATR_XX:  0  0  0  0  0  0  0  0  0  0  0
       OUT:  0  0  0  0  0  0  0  0  0  0  0
  READBACK:  0  0  0  0  0  0  0  0  0  0  0

Configured GPIO values:
[… cut the test results, I can provide them if useful] 

All tests passed!

Done!
</pre>

Please let me now if you need more details, if there are any tests I should run, or if I missed something else…

trungnob, 2018-03-14:
thanks for report I think That is the bug on our side! There will be a fix coming up for this.

closed 2018-03-14 by trungnob

mbr0wn, 2018-03-14:
@fmeynadier To be clear: We didn't close this because we don't see it as an issue, we've closed it because we've duplicated this issue on our internal tracker and are currently working on this.

fmeynadier, 2018-03-14:
Thanks for the quick answer and for this clarification. I suspected this
was the spirit !

Happy bug hunting...

mbr0wn, 2018-03-14:
We have a fix on master (4844f66dccaa71da102b02bba7b8caf8f84a932c) and will apply the same fix on maint too.

fmeynadier, 2018-03-16:
Just to confirm it works now (tried maint). Thanks for you responsiveness !