EttusResearch/uhd #166: EnvironmentError: IOError: Block ctrl (CE_00_Port_10) packet parse error after a random amount of time when using uhd to change transmit frequency

wfdanowitz, 2018-04-02:
On the e310, I have created a UHD program based off of the tx_waveforms.cpp example to use rfnoc siggen to generate samples and send them through the transmitter. If I use set_tx_frequency in the program loop, UHD will successfully change the frequency for a random number of times before failing with a message like:

Error: EnvironmentError: IOError: [0/Radio_0] user_reg_read32() failed: EnvironmentError: IOError: [0/Radio_0] sr_read32() failed: EnvironmentError: IOError: Block ctrl (CE_00_Port_10) packet parse error - EnvironmentError: IOError: Expected packet index: 0  Received index: 1001

Note, the expected packet index is always 0, but the provided "received index" changes from run to run.

```
for (int r = 0; r < radios.size(); r++){
	    freq_iter=freq_iter+1;
	    
	    if (freq_iter == int(84e6/(rate*.8))){freq_iter=0;}
	    boost::this_thread::sleep(boost::posix_time::microseconds(100));
	    radios[r]->set_tx_frequency((base_freq+rate+freq_iter*rate), channel_nums[r]);
	    printf("Freq changed\n");
	  }
```

michael-west, 2018-04-23:
@wfdanowitz Which branch/version of UHD are you using?

michael-west, 2018-07-18:
Closing due to lack of information from user.

@wfdanowitz Please re-open if you are still experiencing the issue, but please provide more detail on how to reproduce the issue.