EttusResearch/uhd #174: Python API will not compile on Windows

jaykuhn, 2018-05-01:
I have easily built the standard interface on Windows. However on Windows I am having problems. I believe they are related to versioning. I have checked out the latest Git code for the python branch. I am using boost 1.58. Some of the things I noticed. 
•	CMake fails because BOOST_PYTHON_FOUND is not found. I have set the recommended variables. I can make it complete by adding BOOT_PYTHON_FOUND true in the CMake GUI. 
•	Transport/zero_copy uses socket.native(.. However in  later versions of boost is depreciated and the call is to native_handle.  However 1.58 versions of the boost binaries only go up to VS version 12.  
•	The maint branch calls native_handle. 
•	I get compiler errors such as:
2> libusb1_base.cpp 
2>C:\Users\Jay\Ettus\git_python_api\uhd\host\lib\transport\libusb1_base.cpp(394): 
error C2600: 'uhd::transport::usb_device_handle::~usb_device_handle' : cannot define a compiler-generated special member function (must be declared in the class first)

•	I have also seen compiler errors complain about definition of a Virtual method. 
•	I want to try a latter VS executable but boost 1.53 binaries at sourceforge only go up to msvc-12.0

Jay

brentstapleton, 2018-05-09:
Hi @jaykuhn, the Python API has recently been updated to include a lot of fixes and features from the UHD master branch. Has the update fixed the issues you saw?

mbr0wn, 2018-07-18:
...also, it builds on Windows. @jaykuhn if you're having issues please get back to us, we've noticed that only certain versions of Python and Boost work on Windows.

jaykuhn, 2018-07-26:
Hi Martin,

Thanks for update.  I have not had time to get back to this until now.   Still having a couple problems . Maybe you could give more details on your exact build environment, boost version etc.

  1.  Updated to “master” on git hub.
  2.  Anaconda2
  3.  Installed ProgramFiles\boost\boost_1_67_0.  Compiled with bootsrap and ./b2
  4.  CMAKE:
     *   Set Boost_INCLUDE_DIR C:/Program Files/boost/boost_1_67_0
     *   Set ENABLE_PYTHON_API
     *   Dependencies for required component LibUHD - Python API not met
     *   I notice “Dependency BOOST_PYTHON_FOUND = “.  Looks like not TRUE
     *   In the GUI Boost_PYTHON_LIBRARY_DEBUG-NOTFOUND (And release
     *   The only place I found anything like DEBUG/RELEASE was: C:/Program Files/boost/boost_1_67_0/tools/build/test/prebuilt/ext/debug/a.h
     *   I am very skeptical this is the correct value but cMake finishes
  5.  Installed Visual Studio 2017 community.
  6.  Open ALL_BUILD and build:
     *   Almost everything fails with message: Done building project "uhd.vcxproj" -- FAILED.
     *   The compile above it has many errors all of form:
     *   : 6>  c:\users\jay\ettus\git_python_api\uhd\host\lib\include\uhdlib\utils\atomic.hpp(34): error C2593: 'operator +' is ambiguous

Thanks,

Jay

jaykuhn, 2018-07-30:
Hi Martin,

I don’t know if you got my comments on Git or not.  The biggest problem I’m having right now is very strange because the Visual compilers are basically saying there are code error.   I’m sure there is something brain dead I have left out.  Maybe there is a header file missing?  Compiler Version?.  I am using ‘master’ branch from git..

More details of my setup on on git.

Thanks,

Jay

1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\transport\super_recv_packet_handler.hpp(626): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\rfnoc\rx_stream_terminator.cpp(112): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\transport\super_recv_packet_handler.hpp(626): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\transport\super_recv_packet_handler.hpp(626): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\transport\super_recv_packet_handler.hpp(626): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\usrp\x300\x300_radio_ctrl_impl.cpp(923): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\usrp\x300\x300_radio_ctrl_impl.cpp(934): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\usrp\x300\x300_dac_ctrl.cpp(212): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\usrp\x300\x300_dac_ctrl.cpp(236): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\transport\super_recv_packet_handler.hpp(626): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\transport\super_recv_packet_handler.hpp(626): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\usrp\n230\n230_stream_manager.cpp(394): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\usrp\common\recv_packet_demuxer_3000.hpp(40): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\utils\atomic.hpp(34): error C2593: 'operator +' is ambiguous
1>c:\users\jay\onedrive\impinj\yukon\ettus\git_python_api\uhd\host\lib\include\uhdlib\utils\atomic.hpp(34): error C2593: 'operator +' is ambiguous

trungnob, 2018-07-30:
hi Jay, 
  I've successfully compiled PythonAPI with following: 
   1/ Visual studio 2017 x64 on Windows 10. 
   2/ Python2.7  x64( not the one come with Visual studio) with numpy installed
   3/ BoostPython 1.65.1 or 1.66 x64 non-static non-debug.
   4/ UHD release build only (non debug build and non static linking)

jaykuhn, 2018-07-31:
Thanks Trung,

The main thing that helped below was compiling release, not debug.  Thanks.   I’m now getting over half my modules build.  However I’m still getting some errors that most likely are due to my configuration not right.   Could you send me your cMake and vxproj files from build so I an compare and see what I have wrong?

Jay

jaykuhn, 2018-08-01:
Tran,

Have you received my e-mails?  Or does this not get through to you?  I would really appreciate getting you configuration files, CmakeChach and vxproj files.  Should be the most straightforward way to confirm my config.

Jay

jaykuhn, 2018-08-02:
Trung,

With a lot of trial and error I am getting close.  I could use more instruction on how to install. 
Current error is:
     ImportError: cannot import name libpyuhd

There is no file or directory called libpyund.  The string does appear inside of of pyuhd.dll

Visual Studio BUILD and INSTAL finished with no error: 

I am using sample code from https://kb.ettus.com/UHD_Python_API

At first import uhd  was not found.  So I added C:/Program Files/UHD/lib/site-packages and now found. 

 C:\Program Files\UHD\bin;

Jay

trungnob, 2018-08-03:
@jaykuhn  sorry for the late reply. I still don't see your email. I think you may send to the wrong address.  
 But I'm glad that you got it compiled! It meant it is 90% done. The installation is a bit of weird first of: 
 1/ Add to system  PATH the path to boost python library (.dll) 
 2/ Add to system PATH the path to libusb.dll ( I assumed you build with usb dll)
 3/ Add to system PATH the path to pyuhd.dll (which is UHD/lib/python/pyuhd.dll) 
 4/ change file name pyuhd.dll -> libpyuhd.pyd
 5/ complete adding system PATH as you said above UHD\bin and UHD\lib\examples\ etc ... to all the UHD executable.

jaykuhn, 2018-08-03:
Hi Trung,

What would be the correct way to get back to you?

Just to update you on more progress but still running into issues.   I had done most of what you suggested except one and I did get a little further.  After that I found:

ImportError: DLL load failed: The specified module could not be found.

I dumped the DLL dependences and tried where one by one. Oddly python27 was not on the path.  I recall that when Anaconda installed it gave a choice to add to path but recommended NO.  Not sure how my environment has been working all these years but adding fixed it.

Now I am at the following error, still at the import of libpyuhd.dll
                ImportError: numpy.core.multiarray failed to import

Any thoughts?

Jay

trungnob, 2018-08-03:
I'm not familiar with Anaconda but that error telling me that the path to numpy's pyd files  some what not correct.
In previous procedures, I actually missed the python path which may not be in the system path.
On my system the numpy's pyd files are at `C:\Program Files\Python27\Lib\Site-packages\numpy\
core\multiarray.pyd`

jaykuhn, 2018-08-04:
I found the problem.  It is related to you suggestion.  I had too many entries in my python paths.  This was in trying to get libpyuhd to resolve.   I removed them and it is ok.   Just to check, I have one path left in my python paths:  \Program Files\UHD\lib\site-packages.   Does that sound correct?

Jay

jaykuhn, 2018-08-06:
Hi Tran,

The problem below was I had some extra paths in my PYTHON_PATHS.    I now have problems with FLASH versions. The probe command returned that the FGPA was 35, but got 13.   I followed the procedure to reflash.  That worked but now I have a residual DDC  error from probe command:

PS C:\Program Files\UHD\bin> .\uhd_usrp_probe.exe --args addr=192.168.10.2
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.1; Boost_106600; UHD_3.13.0.0-0-gf114cfa0
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[INFO] [GPS] No GPSDO found
[WARNING] [UDP] The MTU (1472) is larger than the FastSendDatagramThreshold (1024)!
This will negatively affect the transmit performance.
See the transport application notes for more detail.

[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1307 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1317 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[ERROR] [DDC] Major compat number mismatch for DDC: Expecting 2, got 1.
Error: RuntimeError: FPGA component `DDC' is revision 1 and UHD supports revision 2. Please either upgrade the FPGA image (recommended) or downgrade UHD.

jaykuhn, 2018-08-06:
Hi,

I have 2 boxes.  X300 and N200.    I reconvifgured my network to be 192.168.10.*.   uhd_find_devices finds the X300 at 192.168.1.1,   There is a flashing light at the ethernet connected.

The N200 does not seem to connect.  On power up it flashes the status lights and ends with  D&F constant.  The ethernet port has one solid light.   The find devices command does not find anything.    I have tried a safe boot by holding S2, cycling power, and continue hold S2 until all lights are stable.

Jay

mbr0wn, 2018-08-06:
These questions are better suited to the mailing list and unrelated to the original issue. Please send them to the list.