EttusResearch/uhd #177: Get thread id of threads spawned by UHD?

kkotyk, 2018-05-14:
Is there a mechanism to retrieve UHD thread ids associated with a streamer? The RX threads spawned by UHD do not like being preempted in my application or they go into an unstable state where they continually drop samples. I use isolcpus and taskset with cpu affinity to minimize preemption, but I have no way to distinguish from TX or RX threads. This method seems to be the only feasible way I can use high data rates with multiple usrps(16 N200s) and without the ids I can't do better than basic round robin scheduling.

michael-west, 2018-05-30:
@kkotyk There is no way to get the thread IDs associated with a streamer in UHD at this time.  Marking as an enhancement request.

Please be sure to post questions like these on the usrp-users mailing list (http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com).  Another user may have a solution.

mbr0wn, 2019-02-15:
@kkotyk Since this is a question, I'll close it here, but some additional info:
- Some of our threads do have names, see e.g. this htop output:

[image]

- The RX/TX Streamer always runs in the user's thread, i.e., your application. `benchmark_rate` is an example of how to name the individual threads associated with streamers (again, see picture above).

closed 2019-02-15 by mbr0wn