EttusResearch/uhd #178: fatal error while building UHD

silverninjastar, 2018-05-17:
I was following the [getting started](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux) guide for the USRP B200 and came across the following error while building UHD:
`[ 97%] Building CXX object utils/CMakeFiles/octoclock_firmware_burner.dir/__/lib/utils/ihex.cpp.o
/home/user/workarea-uhd/uhd/host/lib/utils/ihex.cpp:9:10: fatal error: uhdlib/utils/ihex.hpp: No such file or directory
 #include <uhdlib/utils/ihex.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
utils/CMakeFiles/octoclock_firmware_burner.dir/build.make:86: recipe for target 'utils/CMakeFiles/octoclock_firmware_burner.dir/__/lib/utils/ihex.cpp.o' failed
make[2]: *** [utils/CMakeFiles/octoclock_firmware_burner.dir/__/lib/utils/ihex.cpp.o] Error 1
CMakeFiles/Makefile2:3550: recipe for target 'utils/CMakeFiles/octoclock_firmware_burner.dir/all' failed
make[1]: *** [utils/CMakeFiles/octoclock_firmware_burner.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2`

I ran `make test` and it returned with 100% tests passed.

mbr0wn, 2018-05-17:
@silverninjastar We have gotten a small number of people finding this, but we haven't yet been able to reproduce. Can you give us some insight what your setup is? If we can reproduce this, we can probably fix it quite easily.
For a local fix on your end, you can append `-DENABLE_OCTOCLOCK=OFF` to `cmake`.

mbr0wn, 2018-05-17:
And which branch are you building?

silverninjastar, 2018-05-17:
I am using Oracle VM VirtualBox with Ubuntu 17.10.1.  Using branch 'maint'.
I have been following the getting started guide and linked previously and the only thing i believe does not match is that the page said to install libcppunit-1.13-0v5 whereas my version is libcppunit-1.14-0. However, I believe i have followed everything else pretty strictly.

mbr0wn, 2018-05-17:
@silverninjastar Thanks a lot, this is useful information. UHD does not use libcppunit (it is used by GNU Radio), so I doubt that matters.

mbr0wn, 2018-05-17:
@silverninjastar Actually, I think I found the issue. There is a stray reference to ihex.cpp in the `host/utils/CMakeLists.txt`, and in some cases, the `INCLUDE_DIRECTORIES` get resolved such that they don't apply to that file. You can locally fix your maint branch by applying this patch (save to file and run `git apply <filename>`), or simply disable Octoclock as described above. For our next release, we've already fixed this.

```diff
diff --git a/host/utils/CMakeLists.txt b/host/utils/CMakeLists.txt
index 6c91917cc..cea856525 100644
--- a/host/utils/CMakeLists.txt
+++ b/host/utils/CMakeLists.txt
@@ -83,7 +83,6 @@ IF(ENABLE_OCTOCLOCK)
 
     SET(octoclock_burner_sources
         octoclock_firmware_burner.cpp
-        ${CMAKE_SOURCE_DIR}/lib/utils/ihex.cpp
     )
 
     INCLUDE_DIRECTORIES(${CMAKE_SOURCE_DIR}/lib/usrp_clock/octoclock)
```

Since there's nothing we can do right now for the current state of the maint branch, I'm closing this.

Thanks for reporting!

closed 2018-05-17 by mbr0wn