EttusResearch/uhd #181: Legacy mode requires 2-channel DDCs and DUCs

ejk43, 2018-05-17:
Hey all,

Perhaps this is a design choice you've made, but when upgrading to the recent release I've found a few conditionals within the legacy_compat.cpp prevent me from using DDCs and DUCs with just one channel. For example:
- https://github.com/EttusResearch/uhd/blob/maint/host/lib/rfnoc/legacy_compat.cpp#L620
- https://github.com/EttusResearch/uhd/blob/maint/host/lib/rfnoc/legacy_compat.cpp#L899

While it's obviously no problem for larger FPGAs like the X3xx or N3xx series (or the E320), my E310 is seriously resource limited and I've been using single-channel DDCs and DUCs to help fit resources :(

If I know I am only using a single-channel at a time, then I would like to be able to instantiate a 2-channel noc_block_radio_core along side a 1-channel DDC/DUC to pair with the radio channel I'm using on an application-by-application basis (either A:A or A:B). This seems *theoretically* possible, but legacy_compat does not play nice with this scenario any more... Should this use-case be supported?? 

EJ

mbr0wn, 2018-05-18:
@ejk43 That's a good discussion to have. Our philosophy is that `legacy_compat` is for common use cases, and everyone else should use RFNoC API. Question is, what should the behaviour be if there's fewer DDCs than radios?

ejk43, 2018-05-18:
I'm totally on board with that philosophy.. I love when I can just use the USRP sink or USRP source.

To clarify: the scenario I need to support on the E310 would be...
- 1x noc_block_radio_core with 2 channels (ports/datapaths)
- 1x noc_block_ddc with 1 channel
- 1x noc_block_duc with 1 channel
- (other noc_blocks)

So in this specific case, I would not describe it as "fewer DDCs than radios", since the quantity is the same, but rather "fewer channels per DDC than on the radio".

Though now that you mention it, I will also need to support the case where we actually have fewer DDCs than radios, so that's a good discussion to have too. I think I've seen this problem before in a limited scenario.

ejk43, 2018-05-19:
Why not connect Port 2 of the radio block to Port 1 (the only available port) on the DDC?

mbr0wn, 2018-05-29:
In general, I don't think we can come up with something that legacy_compat can handle if channels are not homogeneous. We could do a `min` to determine the number of channels we can handle easily. All cases that are not somewhat trivially handled, people need to start using RFNOC API.

mbr0wn, 2018-05-29:
The purpose of legacy mode is to make an RFNoC device look like a USRP2 (or any other legacy device). Its purpose is not to capture all possible combinations of RFNoC blocks and map them into `multi_usrp` format though. So we've deliberately limited its capabilities.

If you have fewer DDCs than you have radios (let's ignore how they're distributed across blocks), legacy_compat would have to make a decision. But there should be no decision to be made, all connections should b unambiguous.

Also, the connections are done before you set a subdev spec. And they are not changed, either. When you set subdev spec, that needs to apply to existing connections.

So, bottom line, if legacy_compat can't find an easy mapping of the available RFNoC blocks, then it simply gives up. The user can then simply use RFNoC API.

ejk43, 2018-05-30:
Okay, thanks, that makes a lot of sense and I understand better now. Happy to close this issue since it is implementing the intended behavior.

One follow up question here:

>  legacy_compat would have to make a decision. But there should be no decision to be made, all connections should b unambiguous

This makes sense as you have defined legacy mode. Do you see any place for a new feature that could be responsible for searching for available blocks and making those connection decisions on an application-by-application basis, rather than throwing an error? If so, I would be happy to help put this together, since I would really like to be able to instantiate fewer DDCs than radio channels for X300 and eventually E320 applications.

mbr0wn, 2018-06-20:
@ejk43 Yeah, I see place for such a feature. I think we have more examples of more complex configuration requirements, such as the need to do frequency hopping synchronized across a bunch of blocks. Do you have a specific implementation idea?

closed 2018-06-20 by mbr0wn