EttusResearch/uhd #184: Latest UHD-3.11.1.0 static compile w/ boost_1_66_0 and Python API enabled failed

amore-amore, 2018-05-21:
-- ######################################################
-- # UHD enabled components                              
-- ######################################################
--   * LibUHD
--   * LibUHD - C API
--   * Examples
--   * Utils
--   * Tests
--   * USB
--   * B200
--   * USRP2
--   * N230
--   * MPMD
--   * Man Pages
-- 
-- ######################################################
-- # UHD disabled components                             
-- ######################################################
--   * LIBERIO
--   * GPSD
--   * B100
--   * E300
--   * USRP1
--   * X300
--   * OctoClock
--   * Manual
--   * API/Doxygen
-- 
-- Building version: 3.11.0.1-0-unknown
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/build/uhd-3.11.0.1/host/build
[  2%] Built target uhd_rpclib
Scanning dependencies of target uhd_static
[  2%] Building CXX object lib/CMakeFiles/uhd_static.dir/usrp/dboard/magnesium/magnesium_radio_ctrl_impl.cpp.o
In file included from /home/build/uhd-3.11.0.1/host/lib/usrp/dboard/magnesium/magnesium_radio_ctrl_impl.cpp:7:
In file included from /home/build/uhd-3.11.0.1/host/lib/usrp/dboard/magnesium/magnesium_radio_ctrl_impl.hpp:12:
/home/build/uhd-3.11.0.1/host/lib/usrp/dboard/magnesium/magnesium_ad9371_iface.hpp:90:23: error: expected member name or ';' after declaration specifiers
    const std::string _L;
    ~~~~~~~~~~~~~~~~~ ^
/usr/include/ctype.h:46:12: note: expanded from macro '_L'
#define _L      0x02
                ^
1 error generated.
*** Error 1 in . (lib/CMakeFiles/uhd_static.dir/build.make:2974 'lib/CMakeFiles/uhd_static.dir/usrp/dboard/magnesium/magnesium_radio_ctrl_impl.cpp.o')
*** Error 1 in . (CMakeFiles/Makefile2:122 'lib/CMakeFiles/uhd_static.dir/all')
*** Error 1 in /home/build/uhd-3.11.0.1/host/build (Makefile:163 'all')

michael-west, 2018-05-21:
That looks like an issue with a conflict on the definition of _L.  Seems like just a poor choice in a variable name.  If you rename the variable in host/lib/usrp/dboard/magnesium/magnesium_ad9371_iface.hpp, does it build?

amore-amore, 2018-05-22:
After changing to _L --> _LL no macro conflict present. But now I have nullptr and nullptr_t error.

-- ######################################################
-- # UHD enabled components                              
-- ######################################################
--   * LibUHD
--   * LibUHD - C API
--   * Examples
--   * Utils
--   * Tests
--   * USB
--   * B200
--   * USRP2
--   * N230
--   * MPMD
--   * Man Pages
-- 
-- ######################################################
-- # UHD disabled components                             
-- ######################################################
--   * LIBERIO
--   * GPSD
--   * B100
--   * E300
--   * USRP1
--   * X300
--   * OctoClock
--   * Manual
--   * API/Doxygen
-- 
-- Building version: 3.11.1.0-0-unknown
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/build/uhd-3.11.1.0/host/build
[  2%] Built target uhd_rpclib
Scanning dependencies of target uhd_static
[  2%] Building CXX object lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o
In file included from /home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:10:
In file included from /home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_impl.hpp:14:
In file included from /home/build/uhd-3.11.1.0/host/include/uhd/transport/bounded_buffer.hpp:11:
In file included from /home/build/uhd-3.11.1.0/host/include/uhd/transport/bounded_buffer.ipp:14:
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/function.hpp:70:
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/preprocessor/iteration/detail/iter/forward1.hpp:52:
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/detail/function_iterate.hpp:14:
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/detail/maybe_include.hpp:22:
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:138:18: error: called object type 'nullptr_t' is not a function or function pointer
          return (*f)(BOOST_FUNCTION_ARGS);
                 ^~~~
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:925:53: note: in instantiation of member function 'boost::detail::function::function_obj_invoker1<nullptr_t, bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::invoke' requested here
        { { &manager_type::manage }, &invoker_type::invoke };
                                                    ^
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:716:13: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::assign_to<nullptr_t>' requested here
      this->assign_to(f);
            ^
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:1061:5: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::function1<nullptr_t>' requested here
    base_type(f)
    ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:892:13: note: in instantiation of function template specialization 'boost::function<bool (boost::intrusive_ptr<uhd::transport::managed_buffer>)>::function<nullptr_t>' requested here
            NULL);
            ^
/usr/include/sys/_null.h:12:14: note: expanded from macro 'NULL'
#define NULL    nullptr
                ^
1 error generated.
*** Error 1 in . (lib/CMakeFiles/uhd_static.dir/build.make:3454 'lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o': cd /home...)
*** Error 1 in . (CMakeFiles/Makefile2:122 'lib/CMakeFiles/uhd_static.dir/all')
*** Error 1 in /home/build/uhd-3.11.1.0/host/build (Makefile:163 'all')
Running tests...
...
[... 7 lines cut ...]
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/preprocessor/iteration/detail/iter/forward1.hpp:52:
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/detail/function_iterate.hpp:14:
In file included from /home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/detail/maybe_include.hpp:22:
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:138:18: error: called object type 'nullptr_t' is not a function or function pointer
          return (*f)(BOOST_FUNCTION_ARGS);
                 ^~~~
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:925:53: note: in instantiation of member function 'boost::detail::function::function_obj_invoker1<nullptr_t, bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::invoke' requested here
        { { &manager_type::manage }, &invoker_type::invoke };
                                                    ^
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:716:13: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::assign_to<nullptr_t>' requested here
      this->assign_to(f);
            ^
/home/build/uhd-3.11.1.0/host/build/boost/include/boost/function/function_template.hpp:1061:5: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::function1<nullptr_t>' requested here
[... cut ...] ffer>)>::function<nullptr_t>' requested here
            NULL);
            ^
/usr/include/sys/_null.h:12:14: note: expanded from macro 'NULL'
#define NULL    nullptr
                ^
1 error generated.
*** Error 1 in . (lib/CMakeFiles/uhd_static.dir/build.make:3454 'lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o': cd /home...)
*** Error 1 in . (CMakeFiles/Makefile2:122 'lib/CMakeFiles/uhd_static.dir/all')
*** Error 1 in /home/build/uhd-3.11.1.0/host/build (Makefile:163 'all')

michael-west, 2018-05-22:
This looks to be an error in Boost where it is using nullptr_t without it being defined.  According to http://en.cppreference.com/w/cpp/types/nullptr_t, it is defined in <cstddef>.  If the Boost library is using it, it should have included that header, but apparently it does not.  Try adding #include <cstddef> in host/lib/usrp/device3/device3_impl.hpp (before the inclusion of host/include/uhd/transport/bounded_buffer.hpp).

amore-amore, 2018-05-23:
I've done three patches to 3.11.1.0

--- magnesium_ad9371_iface.cpp.old	Wed May 16 22:25:48 2018
+++ magnesium_ad9371_iface.cpp	Tue May 22 09:30:53 2018
@@ -37,9 +37,9 @@
 ) : _rpcc(rpcc)
   , _slot_idx(slot_idx)
   , _rpc_prefix((slot_idx == 0) ? "db_0_" : "db_1_")
-  , _L((slot_idx == 0) ? "AD9371-0" : "AD9371-1")
+  , _LL((slot_idx == 0) ? "AD9371-0" : "AD9371-1")
 {
-    UHD_LOG_TRACE(_L,
+    UHD_LOG_TRACE(_LL,
         "Initialized controls with RPC prefix " << _rpc_prefix <<
         " for slot " << _slot_idx);
 }
@@ -52,7 +52,7 @@
     // Note: This sets the frequency for both channels (1 and 2).
     auto which = _get_which(dir, chan);
     auto actual_freq = request<double>("set_freq", which, freq, false);
-    UHD_LOG_TRACE(_L,
+    UHD_LOG_TRACE(_LL,
             _rpc_prefix << "set_freq returned " << actual_freq);
     return actual_freq;
 }
@@ -64,7 +64,7 @@
 ) {
     auto which = _get_which(dir, chan);
     auto retval = request<double>("set_gain", which, gain);
-    UHD_LOG_TRACE(_L, _rpc_prefix << "set_gain returned " << retval);
+    UHD_LOG_TRACE(_LL, _rpc_prefix << "set_gain returned " << retval);
 
     return retval;
     //return 0.0;
@@ -75,7 +75,7 @@
 {
     auto const which = _get_which(dir, chan);
     auto retval = request<double>("set_bw_filter", which, bandwidth);
-    UHD_LOG_TRACE(_L, _rpc_prefix << "set_bw_filter returned " << retval);
+    UHD_LOG_TRACE(_LL, _rpc_prefix << "set_bw_filter returned " << retval);
     return retval;
 }
 
@@ -85,7 +85,7 @@
 ) {
     auto which = _get_which(dir, chan);
     auto retval = request<double>("get_freq", which);
-    UHD_LOG_TRACE(_L, _rpc_prefix << "get_freq returned " << retval);
+    UHD_LOG_TRACE(_LL, _rpc_prefix << "get_freq returned " << retval);
     return retval;
 }
 
@@ -93,14 +93,14 @@
 {
     auto which = _get_which(dir, chan);
     auto retval = request<double>("get_gain", which);
-    UHD_LOG_TRACE(_L, _rpc_prefix << "get_gain returned " << retval);
+    UHD_LOG_TRACE(_LL, _rpc_prefix << "get_gain returned " << retval);
     return retval;
 }
 
 double magnesium_ad9371_iface::get_bandwidth(const size_t /*chan*/, const direction_t /*dir*/)
 {
     // TODO: implement
-    UHD_LOG_WARNING(_L, "Ignoring attempt to get bandwidth");
+    UHD_LOG_WARNING(_LL, "Ignoring attempt to get bandwidth");
     return 0.0;
 }
 
@@ -111,7 +111,7 @@
     // There is only one LO for 2 channels. Using channel 0 for 'which'
     auto which = _get_which(dir, 0);
     auto retval = request<std::string>("set_lo_source", which, source);
-    UHD_LOG_TRACE(_L, _rpc_prefix << "set_lo_source returned " << retval);
+    UHD_LOG_TRACE(_LL, _rpc_prefix << "set_lo_source returned " << retval);
     return retval;
 }
 
@@ -121,7 +121,7 @@
     // There is only one LO for 2 channels. Using channel 0 for 'which'
     auto which = _get_which(dir, 0);
     auto retval = request<std::string>("get_lo_source", which);
-    UHD_LOG_TRACE(_L, _rpc_prefix << "get_lo_source returned " << retval);
+    UHD_LOG_TRACE(_LL, _rpc_prefix << "get_lo_source returned " << retval);
     return retval;
 }
 
--- magnesium_ad9371_iface.hpp.old	Wed May 16 22:25:48 2018
+++ magnesium_ad9371_iface.hpp	Tue May 22 09:27:06 2018
@@ -70,7 +70,7 @@
     template <typename return_type, typename... Args>
     return_type request(std::string const& func_name, Args&&... args)
     {
-        UHD_LOG_TRACE(_L, "[RPC] Calling " << func_name);
+        UHD_LOG_TRACE(_LL, "[RPC] Calling " << func_name);
         return _rpcc->request_with_token<return_type>(
                 _rpc_prefix + func_name,
                 std::forward<Args>(args)...
@@ -87,7 +87,7 @@
     const std::string _rpc_prefix;
 
     //! Logger prefix
-    const std::string _L;
+    const std::string _LL;
 
 };
 
--- device3_impl.cpp.old	Wed May 23 12:19:24 2018
+++ device3_impl.cpp	Tue May 22 23:05:48 2018
@@ -5,6 +5,7 @@
 // SPDX-License-Identifier: GPL-3.0-or-later
 //
 
+#include <../lib/usrp/device3/device3_impl.hpp>
 #include <uhd/utils/log.hpp>
 #include <uhd/rfnoc/block_ctrl_base.hpp>
 #include <uhdlib/rfnoc/graph_impl.hpp>

amore-amore, 2018-05-23:
Still stuck with compile with third error:

-- ######################################################
-- # UHD enabled components                              
-- ######################################################
--   * LibUHD
--   * LibUHD - C API
--   * Examples
--   * Utils
--   * Tests
--   * USB
--   * B200
--   * USRP2
--   * N230
--   * MPMD
--   * Man Pages
-- 
-- ######################################################
-- # UHD disabled components                             
-- ######################################################
--   * LIBERIO
--   * GPSD
--   * B100
--   * E300
--   * USRP1
--   * X300
--   * OctoClock
--   * Manual
--   * API/Doxygen
-- 
-- Building version: 3.11.1.0-0-unknown
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/build/uhd-3.11.1.0/host/build
[  2%] Built target uhd_rpclib
Scanning dependencies of target uhd_static
[  2%] Building CXX object lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:218:9: error: unknown type name 'endianness_t'
        endianness_t endianness,
        ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:260:23: error: use of undeclared identifier 'ENDIANNESS_BIG'
    if (endianness == ENDIANNESS_BIG) {
                      ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:262:14: error: no member named 'chdr' in namespace 'uhd::transport::vrt'
        vrt::chdr::if_hdr_pack_be(pkt, packet_info);
        ~~~~~^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:268:14: error: no member named 'chdr' in namespace 'uhd::transport::vrt'
        vrt::chdr::if_hdr_pack_le(pkt, packet_info);
        ~~~~~^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:384:23: error: use of undeclared identifier 'device3_impl'
    boost::shared_ptr<device3_impl::async_md_type> async_queue;
                      ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:385:23: error: use of undeclared identifier 'device3_impl'
    boost::shared_ptr<device3_impl::async_md_type> old_async_queue;
                      ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:396:9: error: unknown type name 'endianness_t'
        endianness_t endianness,
        ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:414:27: error: use of undeclared identifier 'ENDIANNESS_BIG'
        if (endianness == ENDIANNESS_BIG)
                          ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:416:18: error: no member named 'chdr' in namespace 'uhd::transport::vrt'
            vrt::chdr::if_hdr_unpack_be(packet_buff, if_packet_info);
            ~~~~~^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:421:18: error: no member named 'chdr' in namespace 'uhd::transport::vrt'
            vrt::chdr::if_hdr_unpack_le(packet_buff, if_packet_info);
            ~~~~~^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:462:6: error: use of undeclared identifier 'device3_impl'
bool device3_impl::recv_async_msg(
     ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:472:6: error: use of undeclared identifier 'device3_impl'
void device3_impl::update_rx_streamers(double /* rate */)
     ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:501:19: error: use of undeclared identifier 'device3_impl'
rx_streamer::sptr device3_impl::get_rx_stream(const stream_args_t &args_)
                  ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:679:6: error: use of undeclared identifier 'device3_impl'
void device3_impl::update_tx_streamers(double /* rate */)
     ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:715:2: error: unknown type name 'both_xports_t'
        both_xports_t _xport;
        ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:716:2: error: unknown type name 'both_xports_t'
        both_xports_t _async_xport;
        ^
/home/build/uhd-3.11.1.0/host/lib/usrp/device3/device3_io_impl.cpp:720:19: error: use of undeclared identifier 'device3_impl'
tx_streamer::sptr device3_impl::get_tx_stream(const uhd::stream_args_t &args_)
                  ^
17 errors generated.
*** Error 1 in . (lib/CMakeFiles/uhd_static.dir/build.make:3454 'lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o': cd /home...)
*** Error 1 in . (CMakeFiles/Makefile2:122 'lib/CMakeFiles/uhd_static.dir/all')
*** Error 1 in /home/build/uhd-3.11.1.0/host/build (Makefile:163 'all')

mbr0wn, 2018-05-29:
Which platform was this on?

amore-amore, 2018-05-30:
It is on OpenBSD 6.3amd64 currently. FYI, UHD 3.9.4 with boost 1.58 successfully builds on OpenBSD 6.2amd64 before.

I can do all the tests for latest UHD on 6.3amd64 with boost 1.66 and make some patches if necessary.

mbr0wn, 2018-05-30:
The third error you're seeing looks like some faulty namespace resolution.... but it still doesn't make sense by itself. In particular, the `vrt` namespace is correctly resolved, but then `chdr` is not found.

amore-amore, 2018-05-31:
Using built-in specs.
COLLECT_GCC=egcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/x86_64-unknown-openbsd6.3/4.9.4/lto-wrapper
Target: x86_64-unknown-openbsd6.3
Configured with: /usr/obj/ports/gcc-4.9.4/gcc-4.9.4/configure --enable-libgcj --without-jar --verbose --program-transform-name='s,^,e,' --disable-nls --with-system-zlib --disable-libmudflap --disable-libgomp --disable-tls --with-gnu-ld --with-gnu-as --enable-threads=posix --enable-wchar_t --with-gmp=/usr/local --enable-languages=c,c++,fortran,objc,java,ada --disable-libstdcxx-pch --enable-cpp --prefix=/usr/local --sysconfdir=/etc --mandir=/usr/local/man --infodir=/usr/local/info --localstatedir=/var --disable-silent-rules --disable-gtk-doc
Thread model: posix
gcc version 4.9.4 (GCC)

amore-amore, 2018-05-31:
I'm using the script below to build UHD 3.11.1 statically with boost 1.66 by egcc (gcc 4.9.4) on OpenBSD 6.3 amd64. The same script was used to build UHD 3.9.4 w/ boost 1.58 OpenBSD 6.2 amd64.

#!/bin/sh -v
# Ettus Research UHD driver for USRP devices

echo "\n### Installing packages..."
pkg_add libusb1 gcc cmake python-2.7.14p1
ln -sf /usr/local/bin/python2.7 /usr/local/bin/python2
ln -sf /usr/local/bin/python2.7 /usr/local/bin/python
pkg_add py-setuptools py-pip
pip install mako docutils requests numpy

UHD_ROOT=$(pwd)
UHD_INSTDIR=/usr/local
BOOST_PREFIX="${UHD_ROOT}/build/boost"
mkdir -p build

#echo "\n### Building boost..."
mkdir -p $BOOST_PREFIX
tar -jxf boost_1_66_0.tar.bz2
tar -zxf boost-patches.tar.gz
cd boost_1_66_0
patch -p0 < ../boost-patches/patch-Jamroot
patch -p0 < ../boost-patches/patch-boost_config_compiler_clang_hpp
patch -p0 < ../boost-patches/patch-boost_config_compiler_gcc_hpp
patch -p0 < ../boost-patches/patch-boost_config_platform_bsd_hpp
patch -p0 < ../boost-patches/patch-boost_functional_hash_detail_limits_hpp
patch -p0 < ../boost-patches/patch-boost_log_detail_event_hpp
patch -p0 < ../boost-patches/patch-boost_predef_other_endian_h
patch -p0 < ../boost-patches/patch-boost_test_impl_execution_monitor_ipp
patch -p0 < ../boost-patches/patch-boost_type_traits_is_convertible_hpp
patch -p0 < ../boost-patches/patch-libs_filesystem_src_operations_cpp
patch -p0 < ../boost-patches/patch-tools_build_src_engine_jam_h
patch -p0 < ../boost-patches/patch-tools_build_src_tools_gcc_jam
echo 'using gcc : : eg++ : <cxxflags>"-fvisibility=hidden -fPIC" <linkflags>"" <archiver>"ar" <striper>"strip" <ranlib>"ranlib" <rc>"" : ;' > user-config.jam
BOOST_CONF="runtime-link=shared threadapi=pthread threading=multi link=static variant=release --layout=tagged --build-type=complete --user-config=user-config.jam -sNO_BZIP2=1"
./bootstrap.sh --without-icu --with-libraries=date_time,filesystem,program_options,regex,system,thread,test,serialization,python
./b2 -d2 -j2 -d1 ${BOOST_CONF} --prefix=$BOOST_PREFIX stage
./b2 -d0 -j4 ${BOOST_CONF} --prefix=$BOOST_PREFIX install

echo "\n### Building UHD..."
cd $UHD_ROOT
patch -p0 < patches/CMakeLists.txt.patch
cd $UHD_ROOT/build
cmake -DCMAKE_POSITION_INDEPENDENT_CODE:BOOL=true -DENABLE_LIBUHD=ON -DENABLE_C_API=ON -DENABLE_PYTHON_API=ON -DENABLE_EXAMPLES=ON -DENABLE_UTILS=ON -DENABLE_MANPAGES=ON -DENABLE_TESTS=ON -DENABLE_MANUAL=OFF -DENABLE_DOXYGEN=OFF -DENABLE_MANPAGES=ON -DENABLE_USB=ON -DENABLE_GPSD=OFF -DENABLE_USRP1=OFF -DENABLE_USRP2=ON -DENABLE_B100=OFF -DENABLE_E100=OFF -DENABLE_E300=OFF -DENABLE_X300=OFF -DENABLE_N230=ON -DENABLE_B200=ON -DENABLE_OCTOCLOCK=OFF -DBoost_NO_SYSTEM_PATHS=ON -DBoost_DEBUG=OFF -DBOOST_ROOT=$BOOST_PREFIX -DBOOST_INCLUDEDIR=$BOOST_PREFIX/include -DBOOST_LIBRARYDIR=$BOOST_PREFIX/lib -DENABLE_STATIC_LIBS=ON -DCMAKE_INSTALL_PREFIX=$UHD_INSTDIR ../
make
make test
make install

echo "\n### UHD driver was installed in /usr/local"

amore-amore, 2018-05-31:
boost 1.66 patches (from OpenBSD's boost 1.66 port tree) have been attached.

amore-amore, 2018-05-31:
Some boost 1.66 patches have been attached from OpenBSD's ports tree.

amore-amore, 2018-05-31:
--- CMakeLists.txt.orig	Thu Mar 29 02:44:29 2018
+++ CMakeLists.txt	Sun May 13 12:02:46 2018
@@ -320,7 +320,8 @@
 SET(Boost_ADDITIONAL_VERSIONS
     "1.46.0" "1.46" "1.47.0" "1.47" "1.48.0" "1.48" "1.48.0" "1.49" "1.50.0" "1.50"
     "1.51.0" "1.51" "1.52.0" "1.52" "1.53.0" "1.53" "1.54.0" "1.54" "1.55.0" "1.55"
-    "1.56.0" "1.56" "1.57" "1.57" "1.58" "1.59" "1.60" "1.61"
+    "1.56.0" "1.56" "1.57" "1.57" "1.58" "1.58.0" "1.59" "1.60" "1.61" "1.62" "1.63"
+    "1.64" "1.65" "1.66" "1.66.0" "1.67" "1.67.0"
 )
 FIND_PACKAGE(Boost ${BOOST_MIN_VERSION} COMPONENTS ${BOOST_REQUIRED_COMPONENTS})

amore-amore, 2018-05-31:
Some boost 1.66 patches have been attached from OpenBSD's ports tree.

$OpenBSD: patch-Jamroot,v 1.6 2018/01/31 21:54:49 rsadowski Exp $
Index: Jamroot
--- Jamroot.orig
+++ Jamroot
@@ -134,8 +134,9 @@ import threadapi-feature ;
 path-constant BOOST_ROOT : . ;
 constant BOOST_VERSION : 1.66.0 ;
 constant BOOST_JAMROOT_MODULE : $(__name__) ;
+constant SO_VERSION : ${SO_VERSION} ;
 
-boostcpp.set-version $(BOOST_VERSION) ;
+boostcpp.set-version $(SO_VERSION) ;
 
 use-project /boost/architecture : libs/config/checks/architecture ;
 
$OpenBSD: patch-boost_config_compiler_clang_hpp,v 1.4 2018/01/31 21:54:49 rsadowski Exp $

boost does not like mixing different compilers at build and compile
time.

We build boost itself with gcc-4.2.1 from base, which doesn't support
the integer scalar type __int128, and this results in a
/usr/local/include/boost/config/user.hpp lacking the "BOOST_HAS_INT128"
define.

When compiling something that depends on boost using a newer compiler
like gcc-4.9 or clang, a code snippet in
/usr/local/include/boost/config/compiler/{clang,gcc}.hpp reactivates
__int128 support, but the code containing typedef declarations
inside /usr/local/include/boost/config/suffix.hpp is never reached,
because /usr/local/include/boost/config/user.hpp lacks "BOOST_HAS_INT128".
Hence, the following errors occur:

/usr/local/include/boost/type_traits/is_integral.hpp:72:1: error: 'int128_type' is not a member of 'boost'
/usr/local/include/boost/type_traits/is_integral.hpp:73:1: error: 'uint128_type' is not a member of 'boost'

As a temporary band-aid, completely disable __int128 support on
OpenBSD.

Index: boost/config/compiler/clang.hpp
--- boost/config/compiler/clang.hpp.orig
+++ boost/config/compiler/clang.hpp
@@ -90,7 +90,7 @@
 #  endif
 #endif
 
-#if defined(__SIZEOF_INT128__) && !defined(BOOST_NVCC_CXX03) && !defined(_MSC_VER)
+#if defined(__SIZEOF_INT128__) && !defined(BOOST_NVCC_CXX03) && !defined(_MSC_VER) && !defined(__OpenBSD__)
 #  define BOOST_HAS_INT128
 #endif
 
@@ -316,7 +316,9 @@
 #endif
 
 // Clang has supported the 'unused' attribute since the first release.
+#if defined(__GNUC__) && (__GNUC__ >= 4) || defined(__clang__)
 #define BOOST_ATTRIBUTE_UNUSED __attribute__((__unused__))
+#endif
 
 // Type aliasing hint.
 #if __has_attribute(__may_alias__)

$OpenBSD: patch-boost_config_compiler_gcc_hpp,v 1.2 2017/12/26 19:18:49 rsadowski Exp $

boost does not like mixing different compilers at build and compile
time.

We build boost itself with gcc-4.2.1 from base, which doesn't support
[... 218 lines cut ...]
+toolset.flags gcc.compile OPTIONS <optimization>speed : ;
 toolset.flags gcc.compile OPTIONS <optimization>space : -Os ;
 
 toolset.flags gcc.compile OPTIONS <inlining>off  : -fno-inline ;
@@ -1189,6 +1189,7 @@ rule link ( targets * : sources * : properties * )
 
 actions link bind LIBRARIES
 {
+    echo "$(CONFIG_COMMAND)" -L"$(LINKPATH)" -Wl,$(RPATH_OPTION:E=-R)$(SPACE)-Wl,$(RPATH) -Wl,-rpath-link$(SPACE)-Wl,"$(RPATH_LINK)" -o "$(<)" $(START-GROUP) "$(>)" "$(LIBRARIES)" $(FINDLIBS-ST-PFX) -l$(FINDLIBS-ST) $(FINDLIBS-SA-PFX) -l$(FINDLIBS-SA) $(END-GROUP) $(OPTIONS) $(USER_OPTIONS)
     "$(CONFIG_COMMAND)" -L"$(LINKPATH)" -Wl,$(RPATH_OPTION:E=-R)$(SPACE)-Wl,$(RPATH) -Wl,-rpath-link$(SPACE)-Wl,"$(RPATH_LINK)" -o "$(<)" $(START-GROUP) "$(>)" "$(LIBRARIES)" $(FINDLIBS-ST-PFX) -l$(FINDLIBS-ST) $(FINDLIBS-SA-PFX) -l$(FINDLIBS-SA) $(END-GROUP) $(OPTIONS) $(USER_OPTIONS)
 }
 
@@ -1199,10 +1200,10 @@ rule link.dll ( targets * : sources * : properties * )
     quote-rpath $(targets) ;
 }
 
-# Differs from 'link' above only by -shared.
+# Differs from 'link' above only by -shared and -fPIC.
 actions link.dll bind LIBRARIES
 {
-    "$(CONFIG_COMMAND)" -L"$(LINKPATH)" -Wl,$(RPATH_OPTION:E=-R)$(SPACE)-Wl,$(RPATH) "$(.IMPLIB-COMMAND)$(<[1])" -o "$(<[-1])" $(HAVE_SONAME)-Wl,$(SONAME_OPTION)$(SPACE)-Wl,$(<[-1]:D=) -shared $(START-GROUP) "$(>)" "$(LIBRARIES)" $(FINDLIBS-ST-PFX) -l$(FINDLIBS-ST) $(FINDLIBS-SA-PFX) -l$(FINDLIBS-SA) $(END-GROUP) $(OPTIONS) $(USER_OPTIONS)
+    "$(CONFIG_COMMAND)" -L"$(LINKPATH)" -Wl,$(RPATH_OPTION:E=-R)$(SPACE)-Wl,$(RPATH) "$(.IMPLIB-COMMAND)$(<[1])" -o "$(<[-1])" $(HAVE_SONAME)-Wl,$(SONAME_OPTION)$(SPACE)-Wl,$(<[-1]:D=) -shared -fPIC $(START-GROUP) "$(>)" "$(LIBRARIES)" $(FINDLIBS-ST-PFX) -l$(FINDLIBS-ST) $(FINDLIBS-SA-PFX) -l$(FINDLIBS-SA) $(END-GROUP) $(OPTIONS) $(USER_OPTIONS)
 }
 
 ###

michaelld, 2019-02-13:
Is this issue still happening with the latest RC1?

amore-amore, 2019-02-14:
Michael, you can close the issue. Not present with latest RC1.

closed 2019-02-17 by michaelld