EttusResearch/uhd #186: UHD and sudo

c-benko, 2018-06-04:
When I run uhd/examples/benchmark_rate with and without sudo, the UHD version changes unexpectedly. 

`
$ sudo ./benchmark_rate
> [INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.1.0-release
`

and without sudo

` 
$ ./bench_markrate
> [INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.12.0.0-9-gd19c7590
`

I am assuming this is coming from confusion about a previously installed version of UHD, but I can't figure out why this is happening. 

System information:
Ubuntu 16.04 LTS
UHD installed with pybombs
device (if relevant) is a Ettus X300. 

I have everything running out of a default prefix and the environment should be set up correctly.

mbr0wn, 2018-06-04:
Try running

    $ ldd ./benchmark_rate
    $ sudo ldd ./benchmark_rate

...maybe it'll point you to a different UHD.

c-benko, 2018-06-04:
Thanks, that did show that sudo was pointing to an old uhd. I simply removed the old libuhd.so file, replaced with the most recent install, and now it works. But, that seems unsatisfying.

sudo was looking for `/usr/lib/x86_64-linux-gnu/libuhd.so` but the regular invocation was looking for `~/prefix/default/lib/libuhd.so`. Is there a reason for this? Anyway I can make sudo use the prefix locations? Did I set something/install something incorrectly?

mbr0wn, 2018-06-04:
You said you were using PyBOMBS right? That uses environment variables to set the environment. When you use sudo, you drop into a different environment, your env vars don't apply.

I *think* if you do `sudo -E`, you don't have that issue.

I'll close this, since it's not a bug in UHD.

closed 2018-06-04 by mbr0wn

c-benko, 2018-06-05:
Thanks. Was the issue cause by a version of UHD that was installed from source (not PyBombs) and not properly removed before PyBombs installed UHD?

mbr0wn, 2018-06-05:
That's possible.