EttusResearch/uhd #187: UHD-3.12.0.0 with boost-1.66.0 static build failed

amore-amore, 2018-06-07:
After patching device3_impl.cpp by adding #include <../lib/usrp/device3/device3_impl.hpp>
build stuck using GCC 4.9.4 on device3_io_impl.cpp with error below.

I can test any changes to make it working.

-- 
-- Configuring the python interpreter...
-- Python interpreter: /usr/local/bin/python
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
fatal: Not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- Could not determine git branch. Probably building from tarball.
fatal: Not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- Using UHD Images Directory: OFF
-- Building Static Libraries: ON
-- Build type not specified: defaulting to release.
-- 
-- Configuring Boost C++ Libraries...
CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:326 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:326 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:326 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:326 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:326 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:326 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
[... 246 lines cut ...]
In file included from /home/build/uhd-3.12.0.0/host/build/boost/include/boost/preprocessor/iteration/detail/iter/forward1.hpp:52:
In file included from /home/build/uhd-3.12.0.0/host/build/boost/include/boost/function/detail/function_iterate.hpp:14:
In file included from /home/build/uhd-3.12.0.0/host/build/boost/include/boost/function/detail/maybe_include.hpp:22:
/home/build/uhd-3.12.0.0/host/build/boost/include/boost/function/function_template.hpp:138:18: error: called object type 'nullptr_t' is not a function or function pointer
          return (*f)(BOOST_FUNCTION_ARGS);
                 ^~~~
/home/build/uhd-3.12.0.0/host/build/boost/include/boost/function/function_template.hpp:925:53: note: in instantiation of member function 'boost::detail::function::function_obj_invoker1<nullptr_t, bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::invoke' requested here
        { { &manager_type::manage }, &invoker_type::invoke };
                                                    ^
/home/build/uhd-3.12.0.0/host/build/boost/include/boost/function/function_template.hpp:716:13: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::assign_to<nullptr_t>' requested here
      this->assign_to(f);
            ^
/home/build/uhd-3.12.0.0/host/build/boost/include/boost/function/function_template.hpp:1061:5: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::function1<nullptr_t>' requested here
    base_type(f)
    ^
/home/build/uhd-3.12.0.0/host/lib/usrp/device3/device3_io_impl.cpp:892:13: note: in instantiation of function template specialization 'boost::function<bool (boost::intrusive_ptr<uhd::transport::managed_buffer>)>::function<nullptr_t>' requested here
            NULL);
            ^
/usr/include/sys/_null.h:12:14: note: expanded from macro 'NULL'
#define NULL    nullptr
                ^
1 error generated.
*** Error 1 in . (lib/CMakeFiles/uhd_static.dir/build.make:3483 'lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o': cd /home...)
*** Error 1 in . (CMakeFiles/Makefile2:122 'lib/CMakeFiles/uhd_static.dir/all')
*** Error 1 in /home/build/uhd-3.12.0.0/host/build (Makefile:163 'all')

amore-amore, 2018-08-17:
The same errors while building uhd-3.13.0.0 with boost-1.66.0
------------------------------------------------------------------------
[  2%] Built target uhd_rpclib
Scanning dependencies of target uhd_static
[  2%] Building CXX object lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o
In file included from /home/build/uhd-3.13.0.0/host/lib/usrp/device3/device3_io_impl.cpp:10:
In file included from /home/build/uhd-3.13.0.0/host/include/../lib/usrp/device3/device3_impl.hpp:15:
In file included from /home/build/uhd-3.13.0.0/host/include/uhd/transport/bounded_buffer.hpp:11:
In file included from /home/build/uhd-3.13.0.0/host/include/uhd/transport/bounded_buffer.ipp:14:
In file included from /home/build/uhd-3.13.0.0/host/build/boost/include/boost/function.hpp:70:
In file included from /home/build/uhd-3.13.0.0/host/build/boost/include/boost/preprocessor/iteration/detail/iter/forward1.hpp:52:
In file included from /home/build/uhd-3.13.0.0/host/build/boost/include/boost/function/detail/function_iterate.hpp:14:
In file included from /home/build/uhd-3.13.0.0/host/build/boost/include/boost/function/detail/maybe_include.hpp:22:
/home/build/uhd-3.13.0.0/host/build/boost/include/boost/function/function_template.hpp:138:18: error: called object type 'nullptr_t' is not a function or function pointer
          return (*f)(BOOST_FUNCTION_ARGS);
                 ^~~~
/home/build/uhd-3.13.0.0/host/build/boost/include/boost/function/function_template.hpp:925:53: note: in instantiation of member function 'boost::detail::function::function_obj_invoker1<nullptr_t, bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::invoke' requested here
        { { &manager_type::manage }, &invoker_type::invoke };
                                                    ^
/home/build/uhd-3.13.0.0/host/build/boost/include/boost/function/function_template.hpp:716:13: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::assign_to<nullptr_t>' requested here
      this->assign_to(f);
            ^
/home/build/uhd-3.13.0.0/host/build/boost/include/boost/function/function_template.hpp:1061:5: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer> >::function1<nullptr_t>' requested here
    base_type(f)
    ^
/home/build/uhd-3.13.0.0/host/lib/usrp/device3/device3_io_impl.cpp:892:13: note: in instantiation of function template specialization 'boost::function<bool (boost::intrusive_ptr<uhd::transport::managed_buffer>)>::function<nullptr_t>' requested here
            NULL);
            ^
/usr/include/sys/_null.h:12:14: note: expanded from macro 'NULL'
#define NULL    nullptr
                ^
1 error generated.
*** Error 1 in . (lib/CMakeFiles/uhd_static.dir/build.make:3483 'lib/CMakeFiles/uhd_static.dir/usrp/device3/device3_io_impl.cpp.o': cd /home...)
*** Error 1 in . (CMakeFiles/Makefile2:122 'lib/CMakeFiles/uhd_static.dir/all')
*** Error 1 in /home/build/uhd-3.13.0.0/host/build (Makefile:163 'all')

mbr0wn, 2019-01-18:
I've submitted a fix for that particular compiler error. I'll keep #223 open to see about static builds.

closed 2019-01-18 by mbr0wn