EttusResearch/uhd #190: make test fails 100% of tests

cjosephson, 2018-06-20:
I'm building the maint branch from source, and the 'make test' stage fails all 39 tests. 'make' completed successfully with no errors or warnings, so I am not sure what could be wrong. I'm running Ubuntu 18.04 LTS with kernel 4.4.0.

Does this have a known resolution, and if not, is there a way to get more verbose output to figure out what is wrong?

mbr0wn, 2018-06-20:
@cjosephson Are you running in a Docker? (the 4.4 kernel suggests so).

Not that it matters. We build UHD multiple times per day and run all the tests, so I'm sure this is something easy to fix. You can run `ctest --output-on-failure` or `ctest -V` to see more output.

I'm guessing they all fail immediately?

cjosephson, 2018-06-20:
If I go to the test directory and do `bash <test_name>.sh` the tests pass. But if I do `./<test_name>.sh` it has the same bad variable name error. Which is weird, because I am using the bash shell so I'm not sure what's wrong with `./`

mbr0wn, 2018-06-20:
Can you paste the content of one of the .sh files? Any one will do.

mbr0wn, 2018-06-20:
When you do `bash $something` it might re-initialize your environment.

cjosephson, 2018-06-20:
the contents of paths_test_test.sh:
```
#!/bin/sh
export PATH=/home/t-cojose/uhd/host/build/tests:$PATH
export LD_LIBRARY_PATH=/home/cojose/uhd/host/build/lib:/home/t-cojose/uhd/host/build/tests:/usr/lib/x86_64-linux-gnu:$LD_LIBRARY_PATH
export UHD_RFNOC_DIR=/home/cojose/uhd/host/include/uhd/rfnoc
paths_test
```

mbr0wn, 2018-06-20:
Hm that doesn't tell us anything.
What does your `env` look like? Does it contain the string `Files/Common`, or maybe a newline?

Also, when you do `bash $something` (I think) you skip the shebang. However, the test shell scripts specify `/bin/sh`, which is a different shell (it's not bash). What if you do `/bin/sh ./<test_name>.sh`?

mbr0wn, 2018-06-20:
In particular, what's the contents of `$PATH` or `$LD_LIBRARY_PATH`?

cjosephson, 2018-06-20:
Oh, I somehow missed that it was `#!/bin/sh` and not `#!/bin/bash`.

If I do `/bin/sh <test_name>.sh` it fails. 

My env doesn't seem to have a Files/Common or a newline

```
SHELL=/bin/bash
TERM=xterm-256color
OLDPWD=/home/cojose/uhd/host/build
USER=cojose
NAME=cojo-desktop
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
HOSTTYPE=x86_64
PATH=/home/cojose/uhd/host/build/tests:/home/cojose/bin:/home/cojose/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
PWD=/home/cojose/uhd/host/build/tests
LANG=en_US.UTF-8
SHLVL=1
HOME=/home/cojose
LOGNAME=cojose
XDG_DATA_DIRS=/usr/local/share:/usr/share:/var/lib/snapd/desktop
LESSOPEN=| /usr/bin/lesspipe %s
LESSCLOSE=/usr/bin/lesspipe %s %s
_=/usr/bin/env
```

cjosephson, 2018-06-20:
$LD_LIBRARY_PATH is empty, which seems weird

mbr0wn, 2018-06-20:
No, that's OK. You normally don't have to specify it if you only have standard paths.

Well, I'm baffled. Maybe you can add some `echo` statements in the `<testname>.sh` to see what the intermediate values are? You can print all the variables that are being exported before and after the `export` statement.

Other than that, we'll probably close this issue if we can't reproduce it. I'd love to know if this is a bug in UHD, though (but you're the first to ever encounter this).

mbr0wn, 2018-06-20:
Another idea: Run

    $ dpkg-reconfigure dash

...and select 'sh' over 'dash'.

cjosephson, 2018-06-20:
the dpkg-reconfigure did it! They all pass now. Must have just been a weird system thing.

closed 2018-06-20 by cjosephson

kprime, 2020-07-31:
comment here just to let u know i've encounter the same issue and dpkg-reconfigure solved.
where i'm usingin Ubuntu 18.04 in WSL, along with UHN v3.15.0.0.