EttusResearch/uhd #192: Transmitting Samples errors

karanjarich, 2018-06-25:
i have installed uhd drivers well and am currently testing usrp n210 the problem is that when i try run this test   _### tx_samples_from_file --freq 915e6 --rate 5e6 --gain 10 usrp_samples.dat_ on the cmd am getting this output

C:\Program Files\UHD\lib\uhd\examples>tx_samples_from_file --freq 915e6 --rate 5e6 --gain 10 usrp_samples.dat
Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.010.001.001-0-unknown

Creating the usrp device with: ...
-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes
-- Creating WSA UDP transport for 192.168.10.2:49156
-- Creating WSA UDP transport for 192.168.10.2:49158
-- Creating WSA UDP transport for 192.168.10.2:49157
-- Creating WSA UDP transport for 192.168.10.2:49159
Error: LookupError: IndexError: multi_usrp::mb_root(18446744073709551615) - invalid vector<T> subscript

michael-west, 2018-07-17:
@karanjarich This may be an old issue on UHD 3.10.1.1 that has since been resolved.  Please try updating UHD to the head of the master branch or the latest UHD release (currently 3.12.0.0) and see if the issue persists.

karanjarich, 2018-07-18:
my current version is 3.12.0.0

michael-west, 2018-07-18:
@karanjarich Thank you.  We will look into it.

karanjarich, 2018-07-18:
welcome Michael could this be the reason as to why my simulated GPS signals
are weak

mbr0wn, 2019-05-22:
@karanjarich No, this is a very binary failure mode. If your device is not found, it won't work at all.

We can't reproduce this, so I'll close this unless you can provide more info to reproduce it. Thanks!

closed 2019-05-22 by mbr0wn