EttusResearch/uhd #193: Python-api ImportError: dynamic module does not define init function (initlibpyuhd)

gallpeti96, 2018-06-29:
I followed the instructions given at Building and Installing UHD from source, but when I try to include uhd to my python script, it says:
Traceback (most recent call last):
  File "/home/peti/PycharmProjects/SpectrumMonitoring/sourcecode.py", line 6, in <module>
    import uhd
  File "/usr/local/lib/python2.7/dist-packages/uhd/__init__.py", line 10, in <module>
    from . import types
  File "/usr/local/lib/python2.7/dist-packages/uhd/types.py", line 10, in <module>
    from . import libpyuhd as lib
ImportError: dynamic module does not define init function (initlibpyuhd)

gallpeti96, 2018-06-29:
I'm using Ubuntu 18.04 LTS and I installed boost by sudo apt-get install libboost-all-dev. It installed version 1.65.1.

guruofquality, 2018-06-30:
@gallpeti96 what is your python interpreter for `/home/peti/PycharmProjects/SpectrumMonitoring/sourcecode.py`? `dynamic module does not define init function` is a common error for mixing python modules with the wrong interpreter (2 vs 3)

gallpeti96, 2018-06-30:
@guruofquality the interpreter is Python 2.7.
If I change it for e.g. Python 3.6 and I copy the uhd library to the python3.6 folder, it gives me the same error: ImportError: dynamic module does not define module export function (PyInit_libpyuhd)

trungnob, 2018-06-30:
@gallpeti96  could you try  to find initlibpyuhd on the libpyuhd what do you see? for example on my machine where ```import uhd``` successfully.  Here what I got : 
```console
$nm -C /usr/local/lib/python2.7/dist-packages/uhd/libpyuhd.so | grep initlibpyuhd
0000000000031230 T initlibpyuhd

$nm -C /usr/local/lib/python2.7/dist-packages/uhd/libpyuhd.so | grep libpyuhd
0000000000031230 T initlibpyuhd
0000000000037850 t init_module_libpyuhd()
```

gallpeti96, 2018-06-30:
@trungnob I got similar results:
```
$ nm -C /usr/local/lib/python2.7/dist-packages/uhd/libpyuhd.so | grep initlibpyuhd
0000000000031ec0 t initlibpyuhd

$ nm -C /usr/local/lib/python2.7/dist-packages/uhd/libpyuhd.so | grep libpyuhd
0000000000031ec0 t initlibpyuhd
0000000000038fb0 t init_module_libpyuhd()
```

trungnob, 2018-06-30:
That is not really the same though. For some reason your initlibpyuhd is hidden! (small `t` instead  capital `T`) Sounds like something wrong during linking this shared object.
  following these issues could help  help https://github.com/boostorg/python/issues/131

gallpeti96, 2018-06-30:
@trungnob thank You very much!! I've found out that, if we have  `-fvisibility=default` somewhere,

> the "init" function of a module is not visible and thus can not be loaded in python ([https://github.com/boostorg/python/issues/173])

I found a line in CMakeList.txt
```
IF(NOT WIN32)
        #only export symbols that are declared to be part of the uhd api (non dll platforms)
        UHD_ADD_OPTIONAL_CXX_COMPILER_FLAG(-fvisibility=hidden HAVE_VISIBILITY_HIDDEN)
        UHD_ADD_OPTIONAL_CXX_COMPILER_FLAG(-fvisibility-inlines-hidden HAVE_VISIBILITY_INLINES_HIDDEN)
ENDIF(NOT WIN32)
```
I changed it to:
```
IF(NOT WIN32)
        #only export symbols that are declared to be part of the uhd api (non dll platforms)
        UHD_ADD_OPTIONAL_CXX_COMPILER_FLAG(-fvisibility=default HAVE_VISIBILITY_DEFAULT)
        UHD_ADD_OPTIONAL_CXX_COMPILER_FLAG(-fvisibility-inlines-default HAVE_VISIBILITY_INLINES_DEFAULT)
    ENDIF(NOT WIN32)
```
and it seems to be, this change in the txt file have solved my problem.

trungnob, 2018-07-01:
I see, I thought about that change too. But It may not be a right solution for us. But it may be a good work around for you now. We will follow this issue and see what is the best solution.

closed 2018-08-03 by brentstapleton

gvanem, 2019-01-07:
I get the same link error on Win32 (MSVC). So this issue should IMHO remain open.
Or is `libpyuhd.dll` not meant for Windows?