EttusResearch/uhd #198: [ERROR] [STREAMER] recv packet demuxer unexpected sid 0xdffa1

Raw1mage, 2018-07-18:
version:
ubuntu 16.04 LTS
kernal 4.8.17 lowlatency

hardware
USRP B210

package:
libuhd 3.12
srsLTE current version

> srsenb

[ERROR] [STREAMER] recv packet demuxer unexpected sid 0xdffa1(random everytime)
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0x27ffbe (random everytime)

dkozel, 2018-07-18:
Hello @Raw1mage Can you confirm the exact version of UHD you are running? It can be found by running `uhd_config_info --version`

Do you encounter this error running the `benchmark_rate` utility included with UHD? If not, this is likely a question better raised with the SRS team.

smarek, 2018-07-26:
@dkozel Hi, experiencing the same issue, same error appears when running `benchmark_rate`

```
> uhd_config_info --print-all
UHD 3.12.0.0-0-unknown
Build date: Mon, 25 Jun 2018 19:02:24 UTC
C compiler: GNU 7.3.0
C++ compiler: GNU 7.3.0
C flags: -DUHD_IMAGES_DIR=OFF -DUHD_RFNOC_ENABLED -DHAVE_CONFIG_H -DUHD_LOG_MIN_LEVEL=1 -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_CONSOLE_COLOR-g -O2 -fdebug-prefix-map=/build/uhd-tiOexT/uhd-3.12.0.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2
C++ flags: -DUHD_IMAGES_DIR=OFF -DUHD_RFNOC_ENABLED -DHAVE_CONFIG_H -DUHD_LOG_MIN_LEVEL=1 -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_CONSOLE_COLOR-g -O2 -fdebug-prefix-map=/build/uhd-tiOexT/uhd-3.12.0.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -fvisibility-inlines-hidden
Enabled components: LibUHD, LibUHD - C API, Examples, Utils, Tests, USB, B100, B200, USRP1, USRP2, X300, N230, MPMD, OctoClock
Install prefix: /usr
Boost version: 1.62
Libusb version: 1.0.24
Package path: /usr
Images directory: /usr/share/uhd/images
ABI version string: 3.12.0
```
benchmark_rate 
```
> /usr/lib/uhd/examples/benchmark_rate --random --rx_rate 1 --tx_rate 1
Using random number of samples in send() and recv() calls.

[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106200; UHD_3.12.0.0-0-unknown
[00:00:00.000003] Creating the usrp device with: ...
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO , Firmware Rev 0.929a
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

[00:00:03.513629] Setting device timestamp to 0...
[WARNING] [MULTI_USRP] The hardware does not support the requested RX sample rate:
Target sample rate: 0.000001 MSps
Actual sample rate: 0.031250 MSps

[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
[WARNING] [MULTI_USRP] The hardware does not support the requested RX sample rate:
Target sample rate: 0.000001 MSps
Actual sample rate: 0.062500 MSps

[ERROR] [STREAMER] recv packet demuxer unexpected sid 0x0
[WARNING] [AD936X] Selected Rx sample rate (0.0625 MHz) is less than
analog frontend filter bandwidth (0.2 MHz).
[00:00:03.921653] Testing receive rate 0.062500 Msps on 1 channels
[WARNING] [MULTI_USRP] The hardware does not support the requested TX sample rate:
Target sample rate: 0.000001 MSps
Actual sample rate: 0.062500 MSps

[WARNING] [MULTI_USRP] The hardware does not support the requested TX sample rate:
Target sample rate: 0.000001 MSps
Actual sample rate: 0.062500 MSps

[D00:00:03.922020] Detected Rx sequence error.
[WARNING] [AD936X] Selected Tx sample rate (0.0625 MHz) is less than
analog frontend filter bandwidth (0.2 MHz).
[00:00:03.945732] Testing transmit rate 0.062500 Msps on 1 channels
D[00:00:03.955622] Detected Rx sequence error.
[00:00:03.955827] Timestamp after overrun recovery ahead of error timestamp! Unable to calculate number of dropped samples.(Delta: -2010440 ticks)
^C
```

aaov, 2018-10-11:
Did anyone answer your question? I've had the same issue too using UHD 3.11.

michael-west, 2018-10-11:
We have found an issue in the reset logic in the FPGA that we believe to be the root cause of this issue and are working to resolve it.  See also #208

natetemple, 2019-07-26:
This issue should be resolved with modern UHD (3.14.1.0+). If it is an outstanding issue for anyone, please reopen this issue or email us at support@ettus.com.

closed 2019-07-26 by natetemple