EttusResearch/uhd #199: Compile error with uhd-3.13.0.0-rc1, boost-1.66, gcc-8.1.1

yarda, 2018-07-20:
Before filing an issue, search for solutions here:
- USRP users mailing list (http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com)
- https://kb.ettus.com/Knowledge_Base
- https://files.ettus.com/manual/

## Issue Description ##
[Describe the issue in detail]
```
/builddir/build/BUILD/uhd-3.12.0.0/host/lib/rfnoc/nocscript/parser.cpp:342:14: error: 'static uhd::rfnoc::nocscript::parser::sptr uhd::rfnoc::nocscript::parser::make(uhd::rfnoc::nocscript::function_table::sptr, uhd::rfnoc::nocscript::expression_variable::type_getter_type, uhd::rfnoc::nocscript::expression_variable::value_getter_type)' causes a section type conflict with 'typename boost::move_upd::enable_defdel_call<U, T, void>::type boost::movelib::default_delete<T>::operator()(U*) const [with U = boost::lexer::detail::equivset; T = boost::lexer::detail::equivset]'
 parser::sptr parser::make(
              ^~~~~~
In file included from /usr/include/boost/move/unique_ptr.hpp:25,
                 from /usr/include/boost/spirit/home/support/detail/lexer/generator.hpp:19,
                 from /usr/include/boost/spirit/home/lex/lexer/lexertl/lexer.hpp:15,
                 from /usr/include/boost/spirit/home/lex/lexer_lexertl.hpp:16,
                 from /usr/include/boost/spirit/include/lex_lexertl.hpp:16,
                 from /builddir/build/BUILD/uhd-3.12.0.0/host/lib/rfnoc/nocscript/parser.cpp:10:
/usr/include/boost/move/default_delete.hpp:195:7: note: 'typename boost::move_upd::enable_defdel_call<U, T, void>::type boost::movelib::default_delete<T>::operator()(U*) const [with U = boost::lexer::detail::equivset; T = boost::lexer::detail::equivset]' was declared here
       operator()(U* ptr) const BOOST_NOEXCEPT
       ^~~~~~~~
```
This error is far from being self-descriptive :)
It happened when recompiling for Fedora Rawhide.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. UHD/FPGA Version/Git Hash, Operation System, Hardware
Fedora Rawhide (f29), x86_64

## Expected Behavior ##
[What you expect to happen]
Build OK

## Actual Behaviour ##
[What happens instead e.g. error message]
Build error

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue
e.g. command lines, GNU Radio flow graph screenshot and .grc etc]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

mbr0wn, 2018-07-21:
@yarda OK I built it in a Docker container and it built just fine. Do you have a way of reproducing this?

yarda, 2018-07-21:
I was able to reproduce the problem on multiple koji builders and also with local builds. It was probably some problem with our Fedora tooling/libraries (maybe gcc, maybe boost). Now it seems the problem is gone. I am not going to invest time to finding which package update fixes it. Closing, sorry for noise.

closed 2018-07-21 by yarda

mbr0wn, 2018-07-21:
Thanks for reporting it though! I would rather spend time investigating than break Fedora.