EttusResearch/uhd #201: complie uhd on Raspberry Pi 3B+ met 'boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'

sovi123, 2018-07-31:
cmake version:     3.5.0
gcc/g++ version: 6.3.0 20170516
liboost version:     1.58.0
Raspberry Pi 3B+ with 2018-06-27-raspbian-stretch.zip
met:
`[ 61%] Linking C executable tx_samples_c
../lib/libuhd.so.3.11：对‘boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const’未定义的引用
../lib/libuhd.so.3.11：对‘boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const’未定义的引用
collect2: error: ld returned 1 exit status
examples/CMakeFiles/tx_samples_c.dir/build.make:107: recipe for target 'examples/tx_samples_c' failed
make[2]: *** [examples/tx_samples_c] Error 1
CMakeFiles/Makefile2:494: recipe for target 'examples/CMakeFiles/tx_samples_c.dir/all' failed
make[1]: *** [examples/CMakeFiles/tx_samples_c.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2`
I checked build/examples/CMakeFiles/tx_samples_c.dir/link.txt:
`/usr/bin/cc  -O3 -DNDEBUG   CMakeFiles/tx_samples_c.dir/tx_samples_c.c.o  -o tx_samples_c -rdynamic ../lib/libuhd.so.3.11 getopt/libgetopt.a -lm -lboost_chrono -lboost_date_time -lboost_filesystem -lboost_program_options -lboost_regex -lboost_system -lboost_unit_test_framework -lboost_serialization -lboost_thread -lboost_atomic -lrt -lpthread -ldl -lusb-1.0 -Wl,-rpath,/home/pi/uhd/host/build/lib: `

mbr0wn, 2018-07-31:
What's the UHD version? Latest master?

sovi123, 2018-08-01:
@mbr0wn Yes,master and 3.10.3 with the same error.refernce[https://github.com/EttusResearch/uhd/issues/87](url).I solved this problem with recompling the libboost library.Thank you for your reply.

closed 2018-08-01 by sovi123