EttusResearch/uhd #204: Replace `time_t` in C API

creinke-sandia, 2018-08-10:
## Issue Description ##
[`time_t`](https://en.cppreference.com/w/c/chrono/time_t) is not a suitable type for a portable C API.  The C standard does not specify what the underlying type will be, aside from that it is arithmetic (or, as of C11, real).  This makes it not necessarily portable.  (Although, surely, no one would make `time_t` be `double complex` ...right?)

In practice, the underlying type might be equivalent to either `int32_t` or `int64_t`, and one of those types should be used in the API rather than `time_t`.

mbr0wn, 2018-08-16:
@creinke-sandia We have a patch for this coming up.

brentstapleton, 2018-08-20:
@creinke-sandia Our patch has been merged internally; it should be publicly available on the master branch soon.

closed 2018-08-20 by brentstapleton