EttusResearch/uhd #207: demuxer unexpected sid error, 3.9.5.LTS

vrolbin, 2018-08-23:
Before filing an issue, search for solutions here:
- USRP users mailing list (http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com)
- https://kb.ettus.com/Knowledge_Base
- https://files.ettus.com/manual/

## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. UHD/FPGA Version/Git Hash, Operation System, Hardware

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue
e.g. command lines, GNU Radio flow graph screenshot and .grc etc]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

vrolbin, 2018-08-23:
I understood that the problem was solved for 3.12 and better versions. 
What about 3.9.5.LTS?  Is the backport available?

michael-west, 2018-08-23:
@vrolbin We are not currently planning to backport the change onto UHD-3.9LTS.

The issue is being closed since it is fixed on the latest version of UHD (currently v3.13.0.2).  If you want to request a backport, please contact support@ettus.com.  Be sure to include the reason why a backport is required instead of just upgrading to the latest UHD.

closed 2018-08-23 by michael-west

vrolbin, 2018-08-26:
Ok, thanks