EttusResearch/uhd #210: Start of Burst (tx_timed_samples.cpp)

glouw, 2018-08-31:
https://github.com/EttusResearch/uhd/blob/4e084337e685204a0142d1dff9e41b4e03d0b582/host/examples/tx_timed_samples.cpp#L77

Should be start_of_burst = true for first packet?

michael-west, 2018-08-31:
@glouw  It should, but it doesn't really matter.  The start_of_burst flag is essentially ignored.  The TX control in the FPGA infers the start of the burst by the presence of the data and does not pay any attention to the flag.

closed 2018-08-31 by michael-west

glouw, 2018-10-11:
That sounds a little dangerous for an open source code base, don't you think? You do after all have forks which rely on this functionality.

michael-west, 2018-10-11:
There is nothing in the UHD or FPGA code that relies on start_of_burst for TX.  Any past code that may have relied on it will be prevented from running due to a compat number mismatch between the UHD and FPGA.

glouw, 2018-10-11:
Yes, but your UHD has forks with their own multi_usrp entry points and custom FPGA images that use these tests _and_ rely on start of burst flags. It would be nice to have it do what it advertises.