EttusResearch/uhd #212: uhd_probe_usrp results in segfault

cjosephson, 2018-09-08:
## Issue Description ##
running uhd_usrp_probe terminates early in a segfault

## Setup Details ##
Using an N210 USRP with an UBX-40 daughterboard on a macbook running gentoo linux 
linux; GNU C++ version 6.4.0; Boost_106500; UHD_003.009.004-0-73cd2e9b
driver version commit 6013a511370b9452020adfc72d7893f1c3bb2963 (HEAD -> master, origin/master, origin/HEAD)

However, this issue is present on UHD-3.10 as well.

## Expected Behavior ##
for USRP device information to be returned without a segfault

## Actual Behaviour ##
cjoseph@gentooAir ~ $ uhd_usrp_probe 
linux; GNU C++ version 6.4.0; Boost_106500; UHD_003.009.004-0-73cd2e9b

-- Opening a USRP2/N-Series device...
-- Current recv frame size: 1472 bytes
-- Current send frame size: 1472 bytes
Segmentation fault

## Steps to reproduce the problem ##
run uhd_usrp_probe

## Additional Information ##
uhd_usrp_probe --init-only does not segfault
uhd_usrp_probe does not segfault on UHD-3.9-LTS, and in fact returns device info... but that device info is erroneous (because 3.9 doesn't support the UBX-40, according to some mailing list chatter)

brentstapleton, 2018-09-29:
Hi @cjosephson, are you still seeing this issue? I don't have your _exact_ setup, but I haven't been able to reproduce the segfault.

Are you using the UHD-3.9LTS branch, or the head of `master`? The output of uhd_usrp_probe indicates that you're using Git hash `73cd2e9b` (which, as far as I can tell, is not in our tree), but you say that you're using Git hash `6013a51`. Are you building UHD from source?

I've tried 3.9, 3.10, and master on an N210+UBX-40, and haven't gotten a seg fault.

michael-west, 2018-10-02:
@cjosephson UHD-3.9LTS does support UBX-40 up to rev C.  The segfault is most likely due to having a revision of the UBX that was not supported in 3.9.4.  Support for UBX rev C was added in 3.9.5, which is why the head of UHD-3.9LTS works and 3.9.4 does not.  Updating to a newer version of UHD should resolve the issue.

We are closing this issue because it does not exist on the head of the UHD-3.9LTS branch.  Thank you for reporting!

closed 2018-10-02 by michael-west