EttusResearch/uhd #213: Boost-python component name is not always 'python3' when using python 3.

smunaut, 2018-09-17:
When enabling python3 in UHD, instead of looking for component 'python' for BOOST, it uses 'python3'. However I think this is something Ubuntu/Debian specific. For Gentoo for instance, the component name remains 'python' and just the component version is 3.xxx

So to build, I had to use this patch :

```diff
diff --cc fpga-src
index ebf5eedfe,c909c784d..000000000
--- a/fpga-src
+++ b/fpga-src
diff --git a/host/CMakeLists.txt b/host/CMakeLists.txt
index 00b9102e4..290ca15a8 100644
--- a/host/CMakeLists.txt
+++ b/host/CMakeLists.txt
@@ -289,7 +289,7 @@ OPTION(ENABLE_PYTHON3 "Enable Python 3. Default is Python 2" OFF)
 SET(BOOST_PYTHON_COMPONENT python)
 SET(BOOST_PYTHON_VERSION 2.7)
 IF(${ENABLE_PYTHON3})
-    SET(BOOST_PYTHON_COMPONENT python3)
+    SET(BOOST_PYTHON_COMPONENT python)
     SET(BOOST_PYTHON_VERSION 3)
 ENDIF()
 
@@ -349,8 +349,8 @@ LINK_DIRECTORIES(${Boost_LIBRARY_DIRS})
 SET(BOOST_PYTHON_FOUND ${Boost_PYTHON_FOUND})
 SET(BOOST_PYTHON_LIBRARY ${Boost_PYTHON_LIBRARY})
 IF(${ENABLE_PYTHON3})
-    SET(BOOST_PYTHON_FOUND ${Boost_PYTHON3_FOUND})
-    SET(BOOST_PYTHON_LIBRARY ${Boost_PYTHON3_LIBRARY})
+    SET(BOOST_PYTHON_FOUND ${Boost_PYTHON_FOUND})
+    SET(BOOST_PYTHON_LIBRARY ${Boost_PYTHON_LIBRARY})
 ENDIF()
 
 MESSAGE(STATUS "Boost include directories: ${Boost_INCLUDE_DIRS}")
```

But obviously this would break ubuntu ... I'm not sure what the best option is here to support both options.

777arc, 2018-11-22:
I was installing uhd into a python3 virtual environment and I had to apply @smunaut 's patch to get cmake to find boost_python as well.  Although I'm on an Ubuntu 18 machine, nothing special. 

Here is what I did in case anyone is curious about getting uhd working in a python3 virtual env:

python3 -m venv pyuhd_install
cd pyuhd_install
. bin/activate
pip install numpy mako requests
mkdir src
cd src
git clone https://github.com/EttusResearch/uhd.git
cd host
mkdir build
cd build
cmake -DENABLE_PYTHON3=ON ../
make -j8
make install

My next step is to set up a way to build the minimal amount of UHD needed to run the python API with a B210

mbr0wn, 2019-02-23:
FYI, closed this because we've dropped the the BP dependency on master and replaced it with PyBind11.

closed 2019-02-23 by mbr0wn

michaelld, 2019-02-23:
Sounds good to me. I think the PyBind11 solution is better.