EttusResearch/uhd #216: Regression when changing DDC sample rate

horvathp, 2018-10-02:
## Issue Description ##
When trying to change the sample rate of the RFNoC-based DDC block under GNU Radio, the DDC hangs. No matter if it is a legacy USRP source or a Device3 DDC. In the former case the flowgraph simply hangs, in the latter case gr-ettus dumps "Timeout on chan0" messages at regular intervals. Changing the output rate of the DDC has worked in UHD 3.9 series.

## Setup Details ##
UHD 3.14.0 6af6ac3
FPGA 454b17c5
Ubuntu 16.04, gnuradio-3.7, gr-ettus master
(same behavior on UHD 3.13)

## Expected Behavior ##
The DDC continues to generate samples with the new output sample rate.

## Actual Behaviour ##
The DDC does not produce any output.

## Steps to reproduce the problem ##
Activate the sample rate callback of the DDC block, for example change the output rate by actuating a GUI checkbox; observe the DDC output, e.g. with a QT GUI Frequency Sink.
[image]

## Additional Information ##
N/A

michael-west, 2018-10-17:
Fixed with fpga commit:  https://github.com/EttusResearch/fpga/commit/4f25ed1b4129c94677b66540894a03a4f80306ea

The manifest will be updated soon so the uhd_images_downloader will download FPGA images with the fix.

closed 2018-10-17 by michael-west

horvathp, 2018-11-01:
This didn't solve the bug for me, it still exists in my setup.