EttusResearch/uhd #222: [ERROR] [B200] Got a ctrl packet with unknown SID

ekagero, 2018-11-09:
OS: ubuntu 17.0.4
Kernel: 4.10.0-42-generic
UHD: 3.14.0.0-165-g1d5c186c (3.14.0)
Openairinterface5g (EPC + eNB on one host)

Error:
eNB producing "[ERROR] [B200] Got a ctrl packet with unknown SID x" with mostly the same SID's. On an identical NUC with UHD 3.14.0.0-110-g6af6ac32 (3.14.0) the problem doesn't persist.

Anybody a hint what to do?

michael-west, 2018-11-12:
@ekagero Thanks for reporting.  I'm a bit surprised because commit [20f06168](https://github.com/EttusResearch/uhddev/commit/20f061689c9eb3024d8579331e5ba790320425f2) was supposed to prevent unexpected sid and LIBUSB_TRANSFER_OVERFLOW errors.  That commit is included in 3.14.0.0-165-g1d5c186c but not in UHD 3.14.0.0-110-g6af6ac32, so I would expect the opposite behavior of what you reported.

What UHD API call is resulting in that error?  Aside from running the entire OAI stack, is there a simpler way to reproduce the error?

zhen1888, 2018-11-19:
Has anybody solved this problem? I met this problem in the same case.

ekagero, 2018-11-21:
Sorry I can't tell anymore since I just reverted back to the old checkout 6af6ac32c30d2dc0efa6eab61e4a3920649e3e62. Will try the latest commit with the next installation and see if it persists.

michael-west, 2018-11-21:
@ekagero @zhen1888 We need a way to reproduce the error that does not involve running the entire OAI stack.  Do either of you have a simple application that reproduces the issue?

asallam, 2018-11-24:
Any solution on this problem. i am facing the same issue as well

drmpeg, 2019-01-21:
@michael-west This looks exactly like the issue I posted on the mailing list last week about a regression with send_frame_size. If you look at the source of openairinterface5g, they're using send_frame_size=15360 in the file targets/ARCH/USRP/USERSPACE/LIB/usrp_lib.cpp.

This issue can be easily reproduced in GNU Radio Companion with just a signal source, UHD sink and setting send_frame_size=15360 in the device address field. Or if you prefer:

uhd_siggen -s 1000000 -f 3429000000 -g 50 -a send_frame_size=15360

The issue appears in v3.13.1.0-rc1 or later. v3.13.0.3-rc1 works fine.

michael-west, 2019-02-04:
Fixed with https://github.com/EttusResearch/fpga/commit/13fdb10fe8d9db271524257963bb990ea0f62138

closed 2019-02-04 by michael-west

mazimkhan, 2022-09-21:
@michael-west is your comment above about frame sizes over 8192 not supported still true after the fix for this issue?
We are seeing this issue occasionally on v3.15.0.0. Our `send_frame_size` is `9232`. What is mapping between `send_frame_size` and max Frame size? Why have you suggested to use 7680 for max frame size 8192?

Besides, the other issue we have that we see this error in SRSENB logs, but this error does not cause the stack process to exit. Is there a config to make library exit on this error, so systemctl can restart it?