EttusResearch/uhd #223: Static build uhd-3.13.1.0-rc1 failed with boost_1_66_0 in the same place as six months ago with previous releases

amore-amore, 2018-11-12:
...
CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:343 (FIND_PACKAGE)

CMake Warning at /usr/local/share/cmake/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/local/share/cmake/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/local/share/cmake/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:343 (FIND_PACKAGE)

-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   chrono
--   date_time
--   filesystem
--   program_options
--   regex
--   system
--   unit_test_framework
--   serialization
--   thread
-- Boost include directories: /home/build/uhd-3.13.1.0-rc1/host/build/boost/include
-- Boost library directories: /home/build/uhd-3.13.1.0-rc1/host/build/boost/lib
-- Boost libraries: /home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_chrono-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_date_time-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_filesystem-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_program_options-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_regex-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_system-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_unit_test_framework-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_serialization-mt.a;/home/build/uhd-3.13.1.0-rc1/host/build/boost/lib/libboost_thread-mt.a
--
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
--
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - found
--
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - "import requests" failed
--
-- Python checking for numpy 1.7 or greater
-- Python checking for numpy 1.7 or greater - found
--
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = TRUE
--   Enabling LibUHD support.
--   Override with -DENABLE_LIBUHD=ON/OFF
--
-- Configuring LibUHD - C API support...
--   Dependency ENABLE_LIBUHD = ON
--   Enabling LibUHD - C API support.
--   Override with -DENABLE_C_API=ON/OFF
--
-- Configuring LibUHD - Python API support...
--   Dependency ENABLE_LIBUHD = ON
--   Dependency BOOST_PYTHON_FOUND = ON
--   Dependency HAVE_PYTHON_MODULE_NUMPY = TRUE
--   Dependency PythonLibs_FOUND = TRUE
--   Enabling LibUHD - Python API support.
--   Override with -DENABLE_PYTHON_API=ON/OFF
--
-- Configuring Examples support...
--   Dependency ENABLE_LIBUHD = ON
--   Enabling Examples support.
--   Override with -DENABLE_EXAMPLES=ON/OFF
--
-- Configuring Utils support...
[... 279 lines cut ...]
30/41 Test #30: fe_conn_test .....................***Failed    0.01 sec
      Start 31: eeprom_c_test
31/41 Test #31: eeprom_c_test ....................***Failed    0.01 sec
      Start 32: error_c_test
32/41 Test #32: error_c_test .....................***Failed    0.01 sec
      Start 33: ranges_c_test
33/41 Test #33: ranges_c_test ....................***Failed    0.01 sec
      Start 34: sensors_c_test
34/41 Test #34: sensors_c_test ...................***Failed    0.01 sec
      Start 35: string_vector_c_test
35/41 Test #35: string_vector_c_test .............***Failed    0.00 sec
      Start 36: subdev_spec_c_test
36/41 Test #36: subdev_spec_c_test ...............***Failed    0.00 sec
      Start 37: nocscript_expr_test
37/41 Test #37: nocscript_expr_test ..............***Failed    0.00 sec
      Start 38: nocscript_ftable_test
38/41 Test #38: nocscript_ftable_test ............***Failed    0.00 sec
      Start 39: nocscript_parser_test
39/41 Test #39: nocscript_parser_test ............***Failed    0.00 sec
      Start 40: config_parser_test
40/41 Test #40: config_parser_test ...............***Failed    0.00 sec
      Start 41: paths_test
41/41 Test #41: paths_test .......................***Failed    0.00 sec

0% tests passed, 41 tests failed out of 41

amore-amore, 2018-11-25:
Any improvements to have this issue fixed?

amore-amore, 2019-02-13:
Fixed in uhd-3.14.0.0-rc1

michaelld, 2019-02-13:
OK, since this issue is fixed I'm closing it.

closed 2019-02-13 by michaelld