EttusResearch/uhd #224: uhd-3.13.1.0-rc1 Python checking for requests 2.0 or greater - "import requests" failed but installed

amore-amore, 2018-11-12:
--
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
--
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - found
--
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - "import requests" failed
--
-- Python checking for numpy 1.7 or greater
-- Python checking for numpy 1.7 or greater - found

# pip install requests
Requirement already satisfied: requests in /usr/local/lib/python2.7/site-packages (2.18.4)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/site-packages (from requests) (2018.4.16)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/site-packages (from requests) (3.0.4)
Requirement already satisfied: idna<2.7,>=2.5 in /usr/local/lib/python2.7/site-packages (from requests) (2.6)
Requirement already satisfied: urllib3<1.23,>=1.21.1 in /usr/local/lib/python2.7/site-packages (from requests) (1.22)

brentstapleton, 2018-11-21:
Hey @amore-amore, are you still seeing this issue? If so, can you verify that the Python interpreter that you're using to build UHD (the CMake variable PYTHON_EXECUTABLE) can find your `requests` package?

amore-amore, 2018-11-25:
The same issue with -DPYTHON_EXECUTABLE=/path/to/system/pyton2.7 still can not find 'requests'.
...
-- Boost libraries:
-- Found PythonInterp: /usr/local/bin/python2.7 (found suitable version "2.7.15", minimum required is "2.7")
--
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
--
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - found
--
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - "import requests" failed
--
-- Python checking for numpy 1.7 or greater
-- Python checking for numpy 1.7 or greater - found
--
-- Configuring LibUHD support...
...

amore-amore, 2019-02-13:
Found when requests has been upgraded up to  'requests 2.21.0'

michaelld, 2019-02-13:
What does `python2.7 -c "import requests; print(requests.__version__)"` return for you? For me, it returns "2.21.0" ... so, this version of requests can be made to work with UHD ...

amore-amore, 2019-02-13:
It works smoothly with requiests 2.21.0. Tested

michaelld, 2019-02-13:
OK so that worked but "requests" was not found properly when configuring UHD ... yes? If so, can you clean out the UHD build directory and run the cmake command from the start & then create a file with the cmake printouts as it does its thing & attach that file here (as a .txt file)? That's a good starting point to us being able to help here!

amore-amore, 2019-02-13:
Requests found by UHD 3.14.0.0 after update to requests version 2.21.0. Previous versions of requests don't work with UHD 3.1x.x.x (simply not found).

michaelld, 2019-02-13:
Ah OK I get what you're getting at: 'requests' prior to 2.21.0 were not found by the UHD cmake install script, but now that you've updated to 2.21.0 this version is being found. Yes?

michaelld, 2019-02-13:
Off-issue reply: "Yes it is" ... OK. so I'm going to go ahead and close this issue since it has been resolved (one way or the other). Thanks for keeping at it here!

closed 2019-02-13 by michaelld