EttusResearch/uhd #226: uhd-3.13.1.0-rc1 dynamic build failed with boost_1_66_0

amore-amore, 2018-11-12:
-- The CXX compiler identification is Clang 5.0.1
-- The C compiler identification is Clang 5.0.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
--
-- Configuring the python interpreter...
-- Found PythonInterp: /usr/local/bin/python2 (found suitable version "2.7.14", minimum required is "2.0")
-- Python interpreter: /usr/local/bin/python2
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Could not determine git branch. Probably building from tarball.
-- Using UHD Images Directory: OFF
-- Build type not specified: defaulting to release.
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
--
-- Configuring Boost C++ Libraries...
-- Looking for optional Boost components...
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   python
-- Looking for required Boost components...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   chrono
--   date_time
--   filesystem
--   program_options
--   regex
--   system
--   unit_test_framework
--   serialization
--   thread
--   atomic
-- Boost include directories: /usr/local/include
-- Boost library directories: /usr/local/lib
-- Boost libraries: /usr/local/lib/libboost_chrono-mt.so.8.0;/usr/local/lib/libboost_date_time-mt.so.8.0;/usr/local/lib/li
bboost_filesystem-mt.so.8.0;/usr/local/lib/libboost_program_options-mt.so.8.0;/usr/local/lib/libboost_regex-mt.so.8.0;/usr
/local/lib/libboost_system-mt.so.8.0;/usr/local/lib/libboost_unit_test_framework-mt.so.8.0;/usr/local/lib/libboost_seriali
zation-mt.so.8.0;/usr/local/lib/libboost_thread-mt.so.8.0;/usr/local/lib/libboost_atomic-mt.so.8.0
-- Looking for log2
-- Looking for log2 - not found
-- Found PythonInterp: /usr/local/bin/python2 (found suitable version "2.7.14", minimum required is "2.7")
-- Found PythonLibs: /usr/local/lib/libpython2.7.so.0.0 (found suitable version "2.7.14", minimum required is "2.7")
--
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
--
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - found
[... 528 lines cut ...]
          return (*f)(BOOST_FUNCTION_ARGS);
                 ^~~~
/usr/local/include/boost/function/function_template.hpp:925:53: note: in instantiation of member function 'boost::detail::function::function_obj_invoker1<nullptr_t, bool,
      boost::intrusive_ptr<uhd::transport::managed_buffer> >::invoke' requested here
        { { &manager_type::manage }, &invoker_type::invoke };
                                                    ^
/usr/local/include/boost/function/function_template.hpp:716:13: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer>
      >::assign_to<nullptr_t>' requested here
      this->assign_to(f);
            ^
/usr/local/include/boost/function/function_template.hpp:1061:5: note: in instantiation of function template specialization 'boost::function1<bool, boost::intrusive_ptr<uhd::transport::managed_buffer>
      >::function1<nullptr_t>' requested here
    base_type(f)
    ^
/home/build/uhd-3.13.1.0-rc1/host/lib/usrp/device3/device3_io_impl.cpp:894:13: note: in instantiation of function template specialization 'boost::function<bool
      (boost::intrusive_ptr<uhd::transport::managed_buffer>)>::function<nullptr_t>' requested here
            NULL);
            ^
/usr/include/sys/_null.h:12:14: note: expanded from macro 'NULL'
#define NULL    nullptr
                ^
1 error generated.
*** Error 1 in . (lib/CMakeFiles/uhd.dir/build.make:2811 'lib/CMakeFiles/uhd.dir/usrp/device3/device3_io_impl.cpp.o': cd /home/build/uhd-3.1...)
*** Error 1 in . (CMakeFiles/Makefile2:120 'lib/CMakeFiles/uhd.dir/all')
*** Error 1 in /home/build/uhd-3.13.1.0-rc1/host/build (Makefile:163 'all')

mbr0wn, 2019-01-18:
We have a fix for the NULL thing in the pipeline. Maybe that'll close this issue out.

michaelld, 2019-02-13:
Is this issue still happening with the recent RC1?

amore-amore, 2019-02-13:
This one can be closed.

amore-amore, 2019-02-13:
Let fix static build. Please see error output provided. Then next to it fix the dynamic build.

amore-amore, 2019-02-13:
I haven't test it with 3.14.0.0-rc1 yet.

mbr0wn, 2019-05-22:
I assume the `NULL` related fix helped. If not, please reopen.

closed 2019-05-22 by mbr0wn