EttusResearch/uhd #228: python bindings are installed in the wrong directory

n-west, 2018-11-21:
## Issue Description ##

I use a multi-stage docker build to get a minimal install. After that I build a development docker that installs lots of other utilities. As a result I wind up having the python bindings installed before python2 is installed from apt (in the second stage development docker). During that apt-install with python, I get the following error:

```
Preparing to unpack .../python2.7-minimal_2.7.12-1ubuntu0~16.04.4_amd64.deb ...
new installation of python2.7-minimal; /usr/lib/python2.7/site-packages is a directory
which is expected a symlink to /usr/local/lib/python2.7/dist-packages.
please find the package shipping files in /usr/lib/python2.7/site-packages and
file a bug report to ship these in /usr/lib/python2.7/dist-packages instead
aborting installation of python2.7-minimal
dpkg: error processing archive /var/cache/apt/archives/python2.7-minimal_2.7.12-1ubuntu0~16.04.4_amd64.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
```

Then the build fails. Checking the bare docker to see what is in site-packages...
```
# ls /usr/lib/python2.7/site-packages/
uhd
```

## Setup Details ##
UHD 3.13.1.0-rc1 with a multi-stage docker build.

## Expected Behavior ##
Installing python should work

## Actual Behaviour ##
Installing python fails

brentstapleton, 2018-11-21:
Hey @n-west, thanks for the report. It does seem that the Python API has been installed to the wrong directory. Where are you getting your UHD .deb file? I thought our `apt`-installed package didn't include the Python API. It also seems wrong to install UHD before your system Python, but that might not be relevant to this issue.

As a workaround, I would suggest building the UHD Python API from source. Or you could see if installing your system Python first clears up the error (although that would probably be masking an underlying issue). If you're building the .deb for the UHD package, you could could also use `UHD_PYTHON_DIR` in the CMake flags to specify an install path- that would probably end up being specific to your setup, however.

A little bit of technical detail:
I took inspiration from [GNU Radio's GRPython.cmake](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/GrPython.cmake) on how to determine the install folder, but it appears that they've recently reverted to hard-coding to [install to `dist-packages`](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/GrPython.cmake#L99). I might be wrong here, but my understanding is that dist-packages is a debian-specific folder, and we should be using `site-packages` for non-debian, Posix systems.

We'll take a look at this, but it's still not clear to me what is the root cause, and the best way forward.

n-west, 2018-12-03:
I did install from source. I think this is an edge case, really... I use a multi-stage docker build to build UHD and then install just UHD (and not build-time dependencies) and it just to happens I did the docker COPY before apt installing python, which triggers this error.

On my end, I have reordered the instructions so it's fine for me. It might be that site-utils is the right place 🤷‍♂️ , but that message does present itself suggesting that maybe it's wrong...

natetemple, 2019-07-26:
@n-west we are going to close out this issue as it seems to be resolved on your end and we haven't been able to reproduce it. If this comes up again, or is a blocker for you, please reopen the issue.

closed 2019-07-26 by natetemple