EttusResearch/uhd #236: README does not explain the required C++ language variant

gdt, 2018-12-29:
I am attempting to update the pkgsrc package for uhd (prompted by the old one failing to build because of boost changes, but that's not the point).  Reading the readme and build instructions, I see that gcc >= 4.8 is needed, but there is no mention of which version of C++ is needed.  It seems like a good guess that it's C++11, because otherwise needing 4.8 would be a bug.  ( pkgsrc attempts to encode the particular flavor for C++ that is needed.)

If people think that one needs a C++11 compiler, it would be nice to simply say that.  Or, if that isn't true, then the requirement for gcc 4.8 should be explained further, as I don't follow it.

brentstapleton, 2019-01-04:
The short, but incomplete, answer to your question is that C++11 is required. For our README, requiring a compiler version is sufficient because we only support a short list of compilers. GCC 4.8 (as well as clang 3.3 and MSVC 2015) supports C++11, which is required to build UHD. However, the requirements to build UHD from source are different from building applications which link against UHD.

Internally, UHD uses many C++11 features. Therefore, when building the UHD library from source, our compiler must support C++11. This is where the compiler version requirement comes from. UHD's build system should handle all of this, assuming your system meets our stated requirements. That is, CMake will call the correct compiler with the correct flags in order to build libuhd.so (or whatever it's named on your OS). This library will then be installed alongside the required headers to use the library.

However, most of our headers do not use C++11 features in them. So any user application that does not want to use C++11 can choose not to, with some minor loss of UHD functionality.

It should also be noted that compiler support for C++11 features isn't perfect. For example, `#include <regex>` is buggy in gcc 4.8. So there are still some concerns past the required version, but for the most part, it should just work.

gdt, 2019-01-04:
Thanks for the explanation.  Specifying a required compiler version isn't really sufficient, since some packaging systems (e.g. pkgsrc) want to specify the language version in control files, and the current README led to what feels like it was a fair question.   (I don't mean to question your requirements for specific versions on top of a requirement for a standard.)

As far as I understand, uhd does work on NetBSD, except for the current build failure which I'll follow up on separately at some point.

mbr0wn, 2019-01-04:
@gdt Thanks for the discussion. I hope our lengthy response was useful. It's great to hear that people are using UHD on NetBSD!
If you need to specific a single number for the C++ version, then C++11 is the one you should use, as @brentstapleton mentioned.

I'll close this issue unless there's additional problems.

closed 2019-01-04 by mbr0wn