EttusResearch/uhd #237: [TwinRX]When I use lo sharing, the 'companion' can work correctly, but by 'external' sometimes has π/2 phase ambiguity

zlinwei, 2019-01-02:
TwinRX lo sharing doesn't work correctly.
A Slot J1 LO2 Export connect to B Slot J2 LO2 Input
A Slot J3 LO1 Export connect to B Slot J4 LO1 Input
```cpp
usrp->set_rx_subdev_spec(string("A:0 A:1 B:0 B:1"));
usrp->set_rx_lo_export(true,"all",0);
usrp->set_rx_lo_source("companion","all",1);
usrp->set_rx_lo_source("external","all",2);
usrp->set_rx_lo_source("external","all",3);
```
I expect a fixed phase difference between them. 
Actually, The phase difference between channel 0 and 1 is invariable, but the phase difference between channel 0 and channel 2 always has two values, the phase difference between channel 0 and channel 3 has two values too. 

Do you known how can I make the phase difference between them always fixed? Thanks so much.

michael-west, 2019-01-02:
@zlinwei Please provide the version of UHD you are using.  TwinRX phase alignment is tested for every release, so it should be working.  The code sample you posted is correct, but it is missing the calls to set the frequency.  There is some path selection logic that is frequency dependent, so the frequency must be set on all channels.

zlinwei, 2019-01-03:
Thanks so much. I am using version 3.13.0.2, I found my problem today, the function "set_time_now" is the cause of this problem.

mbr0wn, 2019-01-03:
Good to hear you found the issue! Yeah, you need to run `set_time_next_pps()`.

closed 2019-01-03 by mbr0wn

zlinwei, 2019-01-03:
Thanks so much.