EttusResearch/uhd #246: Scoped Lock used incorrectly.

taliver, 2019-01-28:
Before filing an issue, search for solutions here:
- USRP users mailing list (http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com)
- https://kb.ettus.com/Knowledge_Base
- https://files.ettus.com/manual/

## Issue Description ##
In 
src/host/lib/usrp/e300/e300_eeprom_manager.cpp

A mutex lock is declared as:
boost::mutex::scoped_lock(_mutex);

Which should actually be:
boost::mutex::scoped_lock  _some_lock_name(_mutex);

This is on the 3.13 branch (at least).

taliver, 2019-01-28:
It's also in : e300_sensor_manager.cpp

Thanks.

michael-west, 2019-01-28:
@taliver  Thanks again.  The issue has been raised to R&D.

mbr0wn, 2019-05-22:
All gone on master.

closed 2019-05-22 by mbr0wn