EttusResearch/uhd #249: Crash in pause and start uhd usrp source in windows

FONQRI, 2019-02-05:
Before filing an issue, search for solutions here:
- USRP users mailing list (http://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com)
- https://kb.ettus.com/Knowledge_Base
- https://files.ettus.com/manual/

## Issue Description ##
I created a simple flowgraph :
https://imgur.com/sakXgjw
and a gui check box :
https://imgur.com/4Ro2Av2
and i assigned it to run in properties options like this :
https://imgur.com/3PbrliP
when i run it with usrp b210 it's working but when i uncheck checkbox and check again it crashes.
https://imgur.com/u7eTcvq
I'm using windows 10 and gnuradio version 3.7.13.4.
(I used checkbox to stop and start the flowgraph)
also when I connect gnuradio blocks from my c++ code and I try to use stop of top_block and start it again it crashes.

## Setup Details ##
UHD/master , windows 10, b210
UHD/rfnoc-devel ,git hash: eec24d7b0442616fdbe9adf6b426959677e67f92, windows 10, b210

## Expected Behavior ##
pause and start work without problem like linux.

## Actual Behaviour ##
No error message it only crashes.

## Steps to reproduce the problem ##
I created a simple flowgraph :
https://imgur.com/sakXgjw
and a gui check box :
https://imgur.com/4Ro2Av2
and i assigned it to run in properties options like this :
https://imgur.com/3PbrliP
when i run it with usrp b210 it's working but when i uncheck checkbox and check again it crashes.
https://imgur.com/u7eTcvq
try to pause and start flowgraph by checking and unchecking runn , QT GUI Checkbox.

FONQRI, 2019-02-25:
I compiled gnuradio in Debug mode and tried to use start and stop and start again (when i'm using uhd source) it crashes. exception is throwing from line 153 of host/lib/transport/libusb1_zero_copy.cpp in 
`function UHD_INLINE typename buffer_type::sptr get_new(const double timeout)` in my version.
can you please check it ? if you need i can give you a simple project.

natetemple, 2019-07-26:
Closing, fixed with https://github.com/EttusResearch/uhd/pull/289

closed 2019-07-26 by natetemple