EttusResearch/uhd #250: OpenBSD build patch uhd-3.14.0.0-rc1

amore-amore, 2019-02-13:
--- pathslib.cpp.old    Wed Feb  6 21:44:00 2019
+++ pathslib.cpp        Wed Feb 13 9:35:38 2019
@@ -9,6 +9,8 @@

 #ifdef BOOST_MSVC
 #  include <windows.h>
+#elif __OpenBSD__
+#  include <glob.h>
 #else
 #  include <wordexp.h>
 #endif
@@ -30,6 +32,16 @@
         return path;
     }
     return std::string(result, result+result_len);
+#elif __OpenBSD__
+    glob_t p;
+    std::string return_value;
+    if (glob(path.c_str(), 0, NULL, &p) == 0 && p.gl_pathc > 0) {
+        return_value = std::string(p.gl_pathv[0]);
+    } else {
+        return_value = path;
+    }
+    globfree(&p);
+    return return_value;
 #else
     wordexp_t p;
     std::string return_value;
@@ -42,4 +54,3 @@
     return return_value;
 #endif
 }

michaelld, 2019-02-13:
Can you either quote your patch so that it comes through "as-is", or attach a text patchfile to this issue that we can download & test directly? The text right now cannot easily be used as a patch. Thx!

amore-amore, 2019-02-13:
Please check your ..mit.edu mail

michaelld, 2019-02-13:
Thx I'm just going to attach the patch here for everyone to see & be able to easily use.

Is this the only change required for OpenBSD that you know of to get the latest UHD to build cleanly?

michaelld, 2019-02-13:
[patch-uhd-pathslib.cpp-openbsd.diff.txt](https://github.com/EttusResearch/uhd/files/2861455/patch-uhd-pathslib.cpp-openbsd.diff.txt)

amore-amore, 2019-02-13:
Yes it is. 3.14.0.0 builds but in the end I have a lot of errors. Can provide output for you.

michaelld, 2019-02-13:
Yes please do provide the output

amore-amore, 2019-02-14:
This ticket can be closed too, if patch has been arrived developers for implementation.

michaelld, 2019-02-14:
Yes we do have the patch now. If you build UHD with dynamic libraries (not static, which is its own separate issue for now), is this patch enough for the build to succeed and "make test" to also succeed? Or is more needed? If this patch is enough, then I'll get the patch in the pipeline internally & close this issue. If not, then I'd suggest augmenting this issue with the full set of patches required.

amore-amore, 2019-02-14:
You can close the issue if you integrated the patch.

Please pay attention to #251 with further errors. I've sent you full build log and build script already. Please check mail.

mbr0wn, 2019-02-15:
@amore-amore This patch uses `glob()`, which doesn't do tilde expansion or the like, which `uhd::path_expandvars()` is supposed to do. Or does `glob.h` on OpenBSD work differently? And why doesn't `wordexp.h` work on OpenBSD? It's POSIX, after all. Did you submit a build failure for this particular issue?

michaelld, 2019-02-17:
@amore-amore We're working internally examining your patch. Truth be told, OpenBSD is not a high priority, so we'll see what comes of this. Thanks again for submitting.

closed 2019-02-17 by michaelld