EttusResearch/uhd #251: UHD 3.14.0.0 static build linking errors on OpenBSD 6.4 amd64 release

amore-amore, 2019-02-14:
...
[ 58%] Building CXX object lib/CMakeFiles/uhd.dir/utils/log_c.cpp.o
[ 58%] Building CXX object lib/CMakeFiles/uhd.dir/utils/thread_priority_c.cpp.o
[ 58%] Building CXX object lib/CMakeFiles/uhd.dir/experts/expert_container.cpp.o
[ 58%] Building CXX object lib/CMakeFiles/uhd.dir/experts/expert_factory.cpp.o
[ 58%] Building CXX object lib/CMakeFiles/uhd.dir/transport/libusb1_control.cpp.o
[ 59%] Building CXX object lib/CMakeFiles/uhd.dir/transport/libusb1_zero_copy.cpp.o
[ 59%] Building CXX object lib/CMakeFiles/uhd.dir/transport/libusb1_base.cpp.o
[ 59%] Building CXX object lib/CMakeFiles/uhd.dir/transport/udp_zero_copy.cpp.o
[ 59%] Building CXX object lib/CMakeFiles/uhd.dir/transport/vrt_if_packet.cpp.o
[ 59%] Building CXX object lib/CMakeFiles/uhd.dir/transport/zero_copy_flow_ctrl.cpp.o
[ 59%] Building CXX object lib/CMakeFiles/uhd.dir/transport/zero_copy_recv_offload.cpp.o
[ 60%] Building CXX object lib/CMakeFiles/uhd.dir/transport/tcp_zero_copy.cpp.o
[ 60%] Building CXX object lib/CMakeFiles/uhd.dir/transport/buffer_pool.cpp.o
[ 60%] Building CXX object lib/CMakeFiles/uhd.dir/transport/if_addrs.cpp.o
[ 60%] Building CXX object lib/CMakeFiles/uhd.dir/transport/udp_simple.cpp.o
[ 60%] Building CXX object lib/CMakeFiles/uhd.dir/transport/chdr.cpp.o
[ 60%] Building CXX object lib/CMakeFiles/uhd.dir/transport/muxed_zero_copy_if.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/uhd.dir/build_info.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/uhd.dir/deprecated.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/uhd.dir/device.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/uhd.dir/device3.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/uhd.dir/image_loader.cpp.o
[ 61%] Building CXX object lib/CMakeFiles/uhd.dir/stream.cpp.o
[ 62%] Building CXX object lib/CMakeFiles/uhd.dir/exception.cpp.o
[ 62%] Building CXX object lib/CMakeFiles/uhd.dir/property_tree.cpp.o
[ 62%] Building CXX object lib/CMakeFiles/uhd.dir/version.cpp.o
[ 62%] Building CXX object lib/CMakeFiles/uhd.dir/error_c.cpp.o
[ 62%] Linking CXX shared library libuhd.so
[ 67%] Built target uhd
Scanning dependencies of target rx_samples_to_file
[ 67%] Building CXX object examples/CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o
[ 67%] Linking CXX executable rx_samples_to_file
/home/build/uhd/uhd-3.14.0.0-rc1/host/build/lib/libuhd.so.3.14: warning: strcpy() is almost always misused, please use strlcpy()
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `boost::detail::lexical_istream_limited_src<char, std::__1::char_traits<char>, true, 29ul>::shl_real_type(double, char*)':
rx_samples_to_file.cpp:(.text._ZN5boost6detail27lexical_istream_limited_srcIcNSt3__111char_traitsIcEELb1ELm29EE13shl_real_typeEdPc[_ZN5boost6detail27lexical_istream_limited_srcIcNSt3__111char_traitsIcEELb1ELm29EE13shl_real_typeEdPc]+0x62): warning: sprintf() is often misused, please use snprintf()
/home/build/uhd/uhd-3.14.0.0-rc1/host/build/lib/libuhd.so.3.14: warning: strcat() is almost always misused, please use strlcat()
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `_main(int, char**)':
rx_samples_to_file.cpp:(.text+0x1862): undefined reference to `boost::program_options::options_description::options_description(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, unsigned int)'
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `boost::program_options::typed_value<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, char>::xparse(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) const':
rx_samples_to_file.cpp:(.text._ZNK5boost15program_options11typed_valueINSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEcE6xparseERNS_3anyERKNS2_6vectorIS8_NS6_IS8_EEEE[_ZNK5boost15program_options11typed_valueINSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEEcE6xparseERNS_3anyERKNS2_6vectorIS8_NS6_IS8_EEEE]+0x86): undefined reference to `boost::program_options::validate(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*, int)'
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `void boost::program_options::validate<unsigned long, char>(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, unsigned long*, long)':
rx_samples_to_file.cpp:(.text._ZN5boost15program_options8validateImcEEvRNS_3anyERKNSt3__16vectorINS4_12basic_stringIT0_NS4_11char_traitsIS7_EENS4_9allocatorIS7_EEEENSA_ISC_EEEEPT_l[_ZN5boost15program_options8validateImcEEvRNS_3anyERKNSt3__16vectorINS4_12basic_stringIT0_NS4_11char_traitsIS7_EENS4_9allocatorIS7_EEEENSA_ISC_EEEEPT_l]+0x131): undefined reference to `boost::program_options::invalid_option_value::invalid_option_value(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `boost::program_options::validation_error::validation_error(boost::program_options::validation_error::kind_t, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int)':
rx_samples_to_file.cpp:(.text._ZN5boost15program_options16validation_errorC2ENS1_6kind_tERKNSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEESB_i[_ZN5boost15program_options16validation_errorC2ENS1_6kind_tERKNSt3__112basic_stringIcNS3_11char_traitsIcEENS3_9allocatorIcEEEESB_i]+0x70): undefined reference to `boost::program_options::error_with_option_name::error_with_option_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int)'
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `void boost::program_options::validate<double, char>(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, double*, long)':
rx_samples_to_file.cpp:(.text._ZN5boost15program_options8validateIdcEEvRNS_3anyERKNSt3__16vectorINS4_12basic_stringIT0_NS4_11char_traitsIS7_EENS4_9allocatorIS7_EEEENSA_ISC_EEEEPT_l[_ZN5boost15program_options8validateIdcEEvRNS_3anyERKNSt3__16vectorINS4_12basic_stringIT0_NS4_11char_traitsIS7_EENS4_9allocatorIS7_EEEENSA_ISC_EEEEPT_l]+0x129): undefined reference to `boost::program_options::invalid_option_value::invalid_option_value(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `boost::program_options::basic_command_line_parser<char>::basic_command_line_parser(int, char const* const*)':
rx_samples_to_file.cpp:(.text._ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc[_ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc]+0xa1): undefined reference to `boost::program_options::detail::cmdline::cmdline(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&)'
CMakeFiles/rx_samples_to_file.dir/rx_samples_to_file.cpp.o: In function `boost::program_options::basic_command_line_parser<char>::extra_parser(boost::function1<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::basic_string<char, std::__1::char_traits<char>, st [... cut ...]
/home/build/uhd/uhd-3.14.0.0-rc1/host/build/boost/lib/libboost_program_options-mt.a(cmdline.o): In function `boost::program_options::detail::cmdline::run()':
cmdline.cpp:(.text+0xe5d): undefined reference to `std::string::empty() const'
cmdline.cpp:(.text+0x1021): undefined reference to `std::string::empty() const'
cmdline.cpp:(.text+0x116f): undefined reference to `std::string::empty() const'
cmdline.cpp:(.text+0x128f): undefined reference to `std::string::operator=(std::string const&)'
cmdline.cpp:(.text+0x12ef): undefined reference to `std::string::size() const'
...

michaelld, 2019-02-17:
Do you have a patch to fix this static linking issue, or are you still working on it? Is the patch from #250 relevant here?

amore-amore, 2019-02-21:
I have no patch to fix this static linking issue. <wordexp.h> and <glob.h> are different according to:
http://pubs.opengroup.org/onlinepubs/009604499/basedefs/glob.h.html
http://pubs.opengroup.org/onlinepubs/009604499/basedefs/wordexp.h.html

It seems something should be fixed in deep UHD driver code to have linking works using <glob.h>.

amore-amore, 2019-02-21:
michaelld, you can reproduce the issue with build script and boots patches I've mailed to you lately.

michaelld, 2019-02-22:
@amore-amore OK :+1: for letting me know. I'll try a static UHD build & see what comes of it. I'm going to go ahead and close this issue and handle it internally within ER. Thanks for your feedback!

closed 2019-02-22 by michaelld

mbr0wn, 2019-02-22:
@amore-amore The purpose of `path_expandvars()`, as the name suggests, is to expand vars such as `$HOME` and `~`. Now if you check `man glob`, it says:

>        The  glob() function searches for all the pathnames matching pattern according to the rules used by the shell (see glob(7)).  No tilde expansion or parameter
>        substitution is done; if you want these, use wordexp(3).

Hence, we can't make this function work with `glob()`.