EttusResearch/uhd #255: RXBasic board not able to sample above nyquist zones in IF mode

JoshSendall, 2019-02-28:
## Issue Description ##
In the RFNoC implementation of IF sampling (i.e. I or Q modes), an initial CIC filter is applied to the channel before the stream leaves the radio block. This limits the frequency range to the range [0, Fs/2]. The RXBasic daughter board, however is supposed to be able to sample above the Nyquist rate.

## Equipment Configuration ##
This issue was noticed when using an X310 with a BasixRX board in I and Q sampling modes.

## Expected Behavior ##
A solution would be to use the "Heterodyne" mode (that the TwinRx uses) to mix the requested center frequency to the base band before the initial CIC filter is applied.

mbr0wn, 2021-09-23:
Hey @JoshSendall, the order of operations in the X310 is:
- Data acquisition (capture interface, sample packaging). This happens in the radio block.
- Next, optionally, we have a DDC block. Here, the order of operations is:
  - DDS-based frequency shift
  - CIC/Halfband filtering + decimation.

The filter comes last, and frequency shifter can be used to put signals into your band of interest.

mbr0wn, 2021-09-23:
Let me show you some examples, using rx_ascii_art_dft. First, I will run at 200 Msps, avoiding the use of DSP blocks. I will use an external signal generator to generate sine waves, and put them into antenna A. That means we're capturing a real signal -- expect an alias at -fc of the signal generator.

## RX rate 200 Msps, fc = 220 MHz

220 MHz is near the top range of the available bandwidth. It should show up at 20 MHz (and thus, at -20 MHz):

    ./examples/rx_ascii_art_dft --args type=x300 --rate 200e6 --freq 0e6 --ant A

[image]

Looks like we can use aliasing. Now, the question is, what happens when we engage the decimation + filters?

## RX rate 50 Msps, fc = 220 MHz

    ./examples/rx_ascii_art_dft --args type=x300 --rate 50e6 --freq 0e6 --ant A

[image]

Even after decimation, the signal is where it used to be.

Now, when we drop the rate, we decrease the size of the usable Nyquist zone. What if we put the center frequency at 120 MHz? It will alias to -80 MHz (and have a mirror image at +80 MHz). Now, we need to engage the frequency shifter.

## RX rate 50 Msps, fc = 120 MHz

    ./examples/rx_ascii_art_dft --args type=x300 --rate 50e6 --freq 110e6 --ant A

My signal is at -80 MHz. I want to keep an offset for nicer pics, so my desired center frequency of the USRP shall be -90 MHz. There's a bit of an idiosyncrasy of UHD here (could be fix-worthy), where it will coerce negative frequencies, so we do the math ourselves and set the center frequency to +110 MHz (which is the same). The signal should show up +10 MHz from the center. Note that rx_ascii_art_dft will label the x-axis with absolute frequencies, so our signal should still show up at -80 MHz (which is the same as our fc=120 MHz, after aliasing). And so it does:

[image]

Note that mirror image at +80 MHz is gone now, filtered away by our decimation filters.

So -- not a bug as far as I can tell.

mbr0wn, 2021-09-23:
BTW, UHD is not entirely clueless about aliasing. See the output here:
```
Setting RX Rate: 50.000000 Msps...
Actual RX Rate: 50.000000 Msps...

Setting RX Freq: 110.000000 MHz...
Actual RX Freq: -90.000000 MHz...
```

closed 2021-09-23 by mbr0wn