EttusResearch/uhd #257: Latest UHD fails to transmit

jdemel, 2019-03-01:
## Issue Description ##
The USRP seems to not transmit anything anymore. This does not happen all the time but most of the time. 

## Setup Details ##
This is the cropped result from `uhd_config_info --print-all` for 2 different systems. Problem occurs with X310 and UBX-160.

Ubuntu 16.04
UHD 3.15.0.git-59-g69c320d7
Build date: Fri, 01 Mar 2019 11:00:28
C compiler: GNU 5.4.0
C++ compiler: GNU 5.4.0
ABI version string: 3.15.0

Ubuntu 18.04
UHD 3.15.0.git-59-g69c320d7
Build date: Fri, 01 Mar 2019 11:48:16
C compiler: GNU 7.3.0
C++ compiler: GNU 7.3.0
ABI version string: 3.15.0

## Expected Behavior ##
I run GQRX in parallel with a HackRF. I expect to see something in the spectrum. But nothing is observed.

## Actual Behaviour ##
No error messages are observed. Just no output. Sometimes no TX network traffic is observed on the host. Sometimes network traffic is observed but no output. Occasionally a test seems to run but rerunning the same test again fails. It does not help to power-cycle the device.

## Steps to reproduce the problem ##
Install particular UHD version.
run `uhd_images_loader`.
run `uhd_image_loader --args ... `
run `tx_waveforms --freq 2.48e9 --rate 12.5e6 --wave-type SQUARE --wave-freq 1e6`
or the GRC flowgraph in `gnuradio/gr-uhd/examples/grc/uhd_dpsk_mod.grc`

Observe spectrum with another device.

mbr0wn, 2019-05-22:
@jdemel I've tried this on various computers, X310s, and even other dboards. I'll leave this open for another bit, but if we can't reproduce this we won't be able to do anything about it.

natetemple, 2019-07-26:
@jdemel This could be a hardware issue. Please email us at support@ettus.com and we can help further debug. Closing this issue.

closed 2019-07-26 by natetemple