EttusResearch/uhd #261: set_rx_freq() with a tune-request does not always apply new frequency on device

Kuurusch, 2019-03-14:
## Issue Description ##
We've encountered, that the set_rx_freq() in a multi-usrp-device does not always apply the desired frequency on the device. But when we request the frequency with get_rx_freq, it is what we wanted to set.

## Setup Details ##
We use 5 x310 to simultaneously recive data over the 10 RX-inputs. The used daughter-boards are UBX-160. Our CPP-application runs on an Ubuntu 18.04 LTS. We use the newest RFNoC image with uhd 3.14.0-rc1 

## Expected Behavior ##
We set with set_rx_freq() the frequencies on all 10 channels one after an other in a for loop. We use tune_request_t. What we expect is, that after this for loop, all channels should be on the desired frequency.

## Actual Behaviour ##
What we encountered is, that from time to time not all channels are set properly and we get no signal on the desired frequency.

Kuurusch, 2019-03-14:
Ok, so I've recognised, that we had on the testing-machine uhd3.14.0-rc3 and the FPGA-image was for 3.14.0-rc1. After I've changed the uhd-version on the machine to rc1, we couldn't reproduce it anymore!

So could it be, that this was the problem?

michael-west, 2019-03-14:
@Kuurusch Yes, this was an issue introduced in 3.14.0.0-rc1 and fixed in 3.14.0.0-rc3.  Since it has been fixed, we will close the issue.  Thank you for reporting!

closed 2019-03-14 by michael-west