EttusResearch/uhd #264: uhd-3.14.0.0 Build problem with boost-1.68.0

barjac, 2019-03-29:
Hi I'm trying to update our (Mageia) package for the next release which now has boost-1.68.0.
Here is a link to a full build log:
http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/uhd-3.14.0.0-1.mga7.src.rpm/build.0.20190329165615.log
A quick search for "undefined reference" will highlight the issues.
 Any help will be much appreciated.

mbr0wn, 2019-03-29:
@barjac We have built with Boost 1.68 before, and have not seen these issues. In your logs, the linker is failing to find almost every single symbol, even Boost ones:
```
/usr/bin/ld: /usr/include/boost/system/error_code.hpp:498: undefined reference to `boost::system::detail::generic_category_ncx()'
/usr/bin/ld: /usr/include/boost/system/error_code.hpp:498: undefined reference to `boost::system::detail::generic_category_ncx()'
/usr/bin/ld: /usr/include/boost/system/error_code.hpp:498: undefined reference to `boost::system::detail::generic_category_ncx()'
/usr/bin/ld: /usr/include/boost/system/error_code.hpp:498: undefined reference to `boost::system::detail::generic_category_ncx()'
```

Are you sure your paths are all set correctly? If only a few symbols were missing, then I'd suspect a different issue, but as it is, I suspect a build setup misconfiguration. Note that I tested Boost 1.68 built from source on Ubuntu, which may be differently configured than yours.

mbr0wn, 2019-04-23:
@barjac We can't reproduce, and we haven't seen this problem before. I'll close this -- but please reopen if you have more information on how we can reproduce this.

closed 2019-04-23 by mbr0wn

barjac, 2019-04-23:
Sorry for my lack of response. I'm not sure just where the problem was, however it's now resolved. Thanks for your efforts!