EttusResearch/uhd #265: Build UHD environment via Docker image

yungshenglu, 2019-04-03:
## Issue Description ##
Hi there, This [link](https://hub.docker.com/r/yungshenglu/uhd-env) my Docker image which has already built UHD software on Ubuntu Linux 16.04 LTS. It may be convenient for building the environment. However, it still needs to transfer the binary file to another computer which connected to USRP for running Tx and Rx.

Please feel free to ask me if you have any problem.

brentstapleton, 2019-04-04:
Hi @yungshenglu, thanks for sharing this! Our Github Issue Track is intended for tracking bugs and other issues in UHD. If you'd like feedback on your Docker image, or would like to show off your work with the community, I'd recommend sending an email to USRP mailing list, usrp-users@lists.ettus.com.

Closing.

closed 2019-04-04 by brentstapleton

yungshenglu, 2019-04-05:
@brentstapleton Thanks for your help.