EttusResearch/uhd #266: UHD 3.14.0.0 does not build with Boost 1.70.0

fxcoudert, 2019-04-15:
## Issue Description ##

Trying to compile UHD 3.14.0.0 on macOS with Boost 1.70.0 (latest version) fails with linker errors, many undefined symbols while trying to link `libuhd.3.14.0.dylib`:

```
  "typeinfo for boost::archive::detail::basic_oserializer", referenced from:
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > in power_container_impl.cpp.o
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > > in power_container_impl.cpp.o
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, std::__1::vector<double, std::__1::allocator<double> > > in power_container_impl.cpp.o
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, std::__1::pair<std::__1::vector<double, std::__1::allocator<double> > const, double> > in power_container_impl.cpp.o
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, std::__1::map<std::__1::vector<double, std::__1::allocator<double> >, double, std::__1::less<std::__1::vector<double, std::__1::allocator<double> > >, std::__1::allocator<std::__1::pair<std::__1::vector<double, std::__1::allocator<double> > const, double> > > > in power_container_impl.cpp.o
  "vtable for boost::detail::thread_data_base", referenced from:
      boost::detail::thread_data_base::thread_data_base() in x300_impl.cpp.o
      boost::detail::thread_data_base::thread_data_base() in tasks.cpp.o
      boost::detail::thread_data_base::thread_data_base() in rpc_client.cpp.o
      boost::detail::thread_data_base::thread_data_base() in zero_copy_recv_offload.cpp.o
      boost::detail::thread_data_base::thread_data_base() in muxed_zero_copy_if.cpp.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
```

We compile (as part of Homebrew https://github.com/Homebrew/homebrew-core/pull/38890) with:

```cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/uhd/3.14.0
.0_1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DENABLE_PYTHON3=ON && make```

The same compilation works with Boost 1.69.0.

michaelld, 2019-04-19:
Thanks for the report. Can you check to see if there are Boost files in `$prefix/lib/cmake"? I'm having issues with the new Boost CMake scripts correctly setting the various Boost CMake variables that we expect for using libraries in the UHD CMake scripts. Headers seem to work.

michaelld, 2019-04-19:
BTW> We need to test this verify, which means installing Boost 1.70.0 on more than 1 system ... which might take a bit so please have patience!

adhorden, 2019-04-21:
I can supply a Docker container that has all the tools and dependencies to compile Boost 1.70.0 from source with UHD from master? 

I have been using this for testing because I have run into the same issues attempting to use GNU Radio with Python 3.7.

michaelld, 2019-04-21:
Thanks for the offer. I think we're about to the point of having Boost 1.70.0 going enough to debug these issues. What it looks like is there's linkage missing for some Boost libraries. Keep updating us on progress if you make any, please & we'll do the same.

michaelld, 2019-04-27:
Can you post the output of the `cmake` command? Also: Did the new Boost install also install its CMake files, probably into `$prefix/lib/cmake/Boost*`? I'm experiencing this issue when I have the new Boost CMake find scripts; the CMake-provided Boost find script seems to work for me && I think these new CMake scripts are new to 1.70.0 ... so, there still might be bugs in them. The cmake debug output is odd with the new scripts, which is why I'm asking for it; just to verify my suspicion here of what's going on. Thx!

fxcoudert, 2019-05-05:
We're running:

```
cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/uhd/3.14.0.0_1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DENABLE_PYTHON3=ON
```

and the output is:

```
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- The C compiler identification is AppleClang 10.0.1.10010046
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- 
-- Configuring the python interpreter...
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7.3", minimum required is "3.0") 
-- Python interpreter: /usr/local/bin/python3
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Python runtime interpreter: /usr/local/bin/python3
-- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
fatal: not a git repository (or any of the parent directories): .git
-- Could not determine git branch. Probably building from tarball.
fatal: not a git repository (or any of the parent directories): .git
-- Using UHD Images Directory: /usr/local/Cellar/uhd/3.14.0.0_1/share/uhd/images
-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
-- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
-- 
-- Configuring Boost C++ Libraries...
-- Looking for optional Boost components...
-- Found Boost 1.70.0 at /usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET OPTIONAL_COMPONENTS python3
-- Found boost_headers 1.70.0 at /usr/local/lib/cmake/boost_headers-1.70.0
-- Boost 1.53 found.
-- Found Boost components:
   python3
-- Looking for required Boost components...
-- Found Boost 1.70.0 at /usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET COMPONENTS chrono;date_time;filesystem;program_options;regex;system;unit_test_framework;serialization;thread
-- Found boost_chrono 1.70.0 at /usr/local/lib/cmake/boost_chrono-1.70.0
--   libboost_chrono-mt.a
--   libboost_chrono.a
-- Adding boost_chrono dependencies: headers
-- Found boost_date_time 1.70.0 at /usr/local/lib/cmake/boost_date_time-1.70.0
--   libboost_date_time-mt.a
--   libboost_date_time.a
-- Adding boost_date_time dependencies: headers
-- Found boost_filesystem 1.70.0 at /usr/local/lib/cmake/boost_filesystem-1.70.0
--   libboost_filesystem-mt.a
--   libboost_filesystem.a
-- Adding boost_filesystem dependencies: headers
-- Found boost_program_options 1.70.0 at /usr/local/lib/cmake/boost_program_options-1.70.0
--   libboost_program_options-mt.a
--   libboost_program_options.a
-- Adding boost_program_options dependencies: headers
-- Found boost_regex 1.70.0 at /usr/local/lib/cmake/boost_regex-1.70.0
--   libboost_regex-mt.a
--   libboost_regex.a
-- Adding boost_regex dependencies: headers
[... 275 lines cut ...]
--   * N230
--   * MPMD
--   * N300
--   * N320
--   * E320
--   * OctoClock
--   * Manual
--   * API/Doxygen
--   * Man Pages
-- 
-- ######################################################
-- # UHD disabled components                             
-- ######################################################
--   * LibUHD - Python API
--   * LIBERIO
--   * GPSD
--   * E300
--   * DPDK
-- 
-- Building version: 3.14.0.0-0-unknown
-- Using install prefix: /usr/local/Cellar/uhd/3.14.0.0_1
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/uhd-20190505-52633-1lh6xql/uhd-3.14.0.0/host/build
```

fxcoudert, 2019-05-05:
Thus full make output is available here: https://gist.github.com/fxcoudert/36a9dc597a952d5c954b15b0656dd70e

michaelld, 2019-05-05:
You have a Boost configuration issue! CMake is finding both Boost 1.70.0 and 1.53 (See the attached chunk of the CMake log you provided for the specifics). You'll note the final 4 lines, where the actual used Boost parameters are listed ... and are empty, which is why the linking is failing. I'd guess that the Boost headers are installed in `${prefix}/include`, which is probably a standard search path for headers and/or is already a directory included for some other project. If the Boost headers were in a nonstandard path, the build would have failed sooner; instead it got to the point of requiring an actual Boost library. You'll note in the link command, there is no `-lboost_*` or a direct `libboost_*` listed ... because of this multi-Boost version conflict.

I _highly_ recommend removing the Boost-provided CMake scripts from `/usr/local/lib/cmake/*oost_*1.70.0` ... those are just not ready for prime time yet (not just me saying this BTW). When I removed those scripts & used the one provided by CMake UHD found the headers & libraries correctly & built cleanly.

Hope this helps!

```
-- Looking for optional Boost components...
-- Found Boost 1.70.0 at /usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET OPTIONAL_COMPONENTS python3
-- Found boost_headers 1.70.0 at /usr/local/lib/cmake/boost_headers-1.70.0
-- Boost 1.53 found.
-- Found Boost components:
   python3
-- Looking for required Boost components...
-- Found Boost 1.70.0 at /usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET COMPONENTS chrono;date_time;filesystem;program_options;regex;system;unit_test_framework;serialization;thread
-- Found boost_chrono 1.70.0 at /usr/local/lib/cmake/boost_chrono-1.70.0
--   libboost_chrono-mt.a
--   libboost_chrono.a
-- Adding boost_chrono dependencies: headers
-- Found boost_date_time 1.70.0 at /usr/local/lib/cmake/boost_date_time-1.70.0
--   libboost_date_time-mt.a
--   libboost_date_time.a
-- Adding boost_date_time dependencies: headers
-- Found boost_filesystem 1.70.0 at /usr/local/lib/cmake/boost_filesystem-1.70.0
--   libboost_filesystem-mt.a
--   libboost_filesystem.a
-- Adding boost_filesystem dependencies: headers
-- Found boost_program_options 1.70.0 at /usr/local/lib/cmake/boost_program_options-1.70.0
--   libboost_program_options-mt.a
--   libboost_program_options.a
-- Adding boost_program_options dependencies: headers
-- Found boost_regex 1.70.0 at /usr/local/lib/cmake/boost_regex-1.70.0
--   libboost_regex-mt.a
--   libboost_regex.a
-- Adding boost_regex dependencies: headers
-- Found boost_system 1.70.0 at /usr/local/lib/cmake/boost_system-1.70.0
--   libboost_system-mt.a
--   libboost_system.a
-- Adding boost_system dependencies: headers
-- Found boost_unit_test_framework 1.70.0 at /usr/local/lib/cmake/boost_unit_test_framework-1.70.0
--   libboost_unit_test_framework-mt.a
--   libboost_unit_test_framework.a
-- Adding boost_unit_test_framework dependencies: headers
-- Found boost_serialization 1.70.0 at /usr/local/lib/cmake/boost_serialization-1.70.0
--   libboost_serialization-mt.a
--   libboost_serialization.a
-- Adding boost_serialization dependencies: headers
-- Found boost_thread 1.70.0 at /usr/local/lib/cmake/boost_thread-1.70.0
--   libboost_thread-mt.a
-- Adding boost_thread dependencies: headers
-- Boost 1.53 found.
-- Found Boost components:
   chrono;date_time;filesystem;program_options;regex;system;unit_test_framework;serialization;thread
-- Boost include directories: 
-- Boost library directories: 
-- Boost libraries: 
```

fxcoudert, 2019-05-06:
- There is no boost 1.53 on the system, only boost 1.70.0.
- We will be shipping boost as it is, with its `cmake` files.

michaelld, 2019-05-06:
If CMake is reporting "Boost 1.53 found", then at a minimum there is -something- installed on your system here that was found by CMake & is reporting Boost 1.53 installed -- it might not be actually installed, but something is reporting it that way.

michaelld, 2019-05-06:
You can certainly choose to ship the Boost CMake scripts, but they aren't ready for being used yet in my testing because multiple projects I've tested don't work with them ... they do not provide the variables GR CMake scripts expect, and it's not clear to me how to fix the issue. Most importantly, by just not using them (just install Boost with the correct flag, and/or "rm -rf" them once installed) and instead using the CMake provided Boost script this all works out of the box. Hence, I doubt we'll be addressing this issue any time soon as there's a viable current option.

fedya, 2019-05-12:
Same here.
Reproducible build list
https://abf.openmandriva.org/build_lists/508078
error:
`/lib/libuhd.so.3.14.0: error: undefined reference to 'boost::chrono::steady_clock::now()'`

michaelld, 2019-05-12:
@fedya A couple points:
(1) I don't know how you do this, but adding in verbosity ("VERBOSE=ON") to the "make" command would be really useful when providing an error log. I can see all the linkage errors, but I can't see the actual link command. I'm guessing it's void of any actual Boost library linkage, because ...

(2) As already stated: The Boost-provided CMake scripts do not play nicely with some projects, including but certainly not limited to UHD and GNU Radio. I've spent a little time trying to read through these scripts & it is not clear to me what's going on nor how to fix them. The UHD CMake scripts work fine with the CMake-provided Boost finding scripts, so it seems as though they are not drop-in replacements for one another. I'd advise you to either not install those Boost CMake scripts, to "rm -rf" them for the current Boost version; hopefully they will work better in the next Boost release.

michaelld, 2019-05-12:
To be clear: When I install those Boost CMake scripts, I get the exact same errors. The issue is easy to replicate; what's not simple is how to create a fix to the UHD / GR CMake scripts so that they will work with both the Boost CMake scripts and the CMake Boost script. Maybe someone has figured this out & there's a patch or 2 available? I'll admit I haven't looked much as there are more important issues to deal with at the moment.

trungnob, 2019-05-13:
@fedya  the error log indicate some sort of confusion during CMake trying to find boost. 
For our UHD we use hint to find boost the hint is at https://github.com/EttusResearch/uhd/blob/master/host/CMakeLists.txt#L313. Without looking at the system, I don't know what is the main reason. Here are some hypothesis: 
  - There are other boost libs that are inside the system and using different type of hint for example Boost_ROOT or Boost Cmake. Look at this link to see https://cmake.org/cmake/help/v3.14/module/FindBoost.html. Could you try to see if there's any hint set in your environment vars. 
BOOST_ROOT or (BOOSTROOT). Or try to set Boost_NO_BOOST_CMAKE to ON to disable the search for boost-cmake.

michaelld, 2019-05-13:
@trungnob The CMake "FindBoost.cmake" script isn't being used. Boost 1.70.0 provides their own CMake find scripts, which is, ultimately, what's causing the issue.

trungnob, 2019-05-13:
humm .that's odd. Just for another record, I was able to build UHD master on Windows 10 x64 with cmake 3.14.3 and Boost 1.70.0 (binary distribution package)  and Visual Studio 2019.

michaelld, 2019-05-13:
If someone has Boost 1.70.0 installed and can build UHD and GR without modification to any of these, then they are not using the Boost-provided CMake scripts; they are using the FindBoost.cmake script. Regardless of OS.

fedya, 2019-05-13:
@trungnob i'm trying to build 3.14.0.0, not master branch.
But let me try...

fedya, 2019-05-13:
Same with master branch
```
[ 92%] Linking CXX shared library libuhd.so
[ 92%] Built target uhd
Scanning dependencies of target uhd_config_info
Scanning dependencies of target uhd_image_loader
Scanning dependencies of target usrp_burn_db_eeprom
Scanning dependencies of target fx2_init_eeprom
Scanning dependencies of target octoclock_burn_eeprom
Scanning dependencies of target usrp_burn_mb_eeprom
Scanning dependencies of target uhd_find_devices
Scanning dependencies of target converter_benchmark
Scanning dependencies of target uhd_cal_rx_iq_balance
Scanning dependencies of target uhd_usrp_probe
Scanning dependencies of target b2xx_fx3_utils
Scanning dependencies of target uhd_cal_tx_dc_offset
Scanning dependencies of target uhd_cal_tx_iq_balance
Scanning dependencies of target query_gpsdo_sensors
Scanning dependencies of target responder
Scanning dependencies of target pyuhd
[ 93%] Building CXX object utils/CMakeFiles/uhd_image_loader.dir/uhd_image_loader.cpp.o
[ 94%] Linking CXX executable uhd_image_loader
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::chrono::steady_clock::now()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::this_thread::interruption_requested()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::thread::thread()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'typeinfo for boost::archive::detail::basic_iserializer'
../lib/libuhd.so.3.15.0: error: undefined reference to 'typeinfo for boost::serialization::typeid_system::extended_type_info_typeid_0'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::extended_type_info::key_unregister() const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::extended_type_info::key_register() const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::is_equal(boost::serialization::extended_type_info const&) const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::is_less_than(boost::serialization::extended_type_info const&) const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::detail::basic_iserializer::~basic_iserializer()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::detail::basic_iserializer::basic_iserializer(boost::serialization::extended_type_info const&)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::detail::basic_iarchive::load_object(void*, boost::archive::detail::basic_iserializer const&)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::text_iarchive_impl<boost::archive::text_iarchive>::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::~extended_type_info_typeid_0()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::extended_type_info_typeid_0(char const*)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::type_unregister()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::type_register(std::type_info const&)'

```

michaelld, 2019-05-13:
FYI that -regardless- of the version of UHD you try to build, whether 3.9LTS, the latest release, any interim release, the current GIT master HEAD, or any prior GIT commit:

==> If you have the Boost CMake scripts installed then the build will fail due to this linking error. <==

michaelld, 2019-05-13:
If you can somehow do "make VERBOSE=ON", then you'll see the link command has no Boost libraries included anywhere in it. The issue is that the return variables from the Boost CMake find scripts and from the CMake Boost find script are different, somehow, and UHD's (and virtually -all- projects out there that use CMake and Boost) CMake scripts are tailored for the latter (the CMake Boost find script), not the former (the Boost CMake find scripts). We are working internally to try to figure out "the right thing to do". Please be patient.

fedya, 2019-05-13:
verbosed build
```
[ 94%] Linking CXX executable uhd_image_loader
cd /home/omv/wspr/wsjtx/uhd/BUILD/uhd-3.14.0.0-20190513/host/build/utils && /usr/bin/cmake -E cmake_link_script CMakeFiles/uhd_image_loader.dir/link.txt --verbose=ON
/usr/bin/c++   -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG   CMakeFiles/uhd_image_loader.dir/uhd_image_loader.cpp.o  -o uhd_image_loader -Wl,-rpath,/home/omv/wspr/wsjtx/uhd/BUILD/uhd-3.14.0.0-20190513/host/build/lib: ../lib/libuhd.so.3.15.0 -lpthread -lrt -ldl /usr/lib64/libusb-1.0.so
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::chrono::steady_clock::now()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::this_thread::interruption_requested()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::thread::thread()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'typeinfo for boost::archive::detail::basic_iserializer'
../lib/libuhd.so.3.15.0: error: undefined reference to 'typeinfo for boost::serialization::typeid_system::extended_type_info_typeid_0'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::extended_type_info::key_unregister() const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::extended_type_info::key_register() const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::is_equal(boost::serialization::extended_type_info const&) const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::is_less_than(boost::serialization::extended_type_info const&) const'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::detail::basic_iserializer::~basic_iserializer()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::detail::basic_iserializer::basic_iserializer(boost::serialization::extended_type_info const&)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::detail::basic_iarchive::load_object(void*, boost::archive::detail::basic_iserializer const&)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::archive::text_iarchive_impl<boost::archive::text_iarchive>::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::~extended_type_info_typeid_0()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::extended_type_info_typeid_0(char const*)'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::type_unregister()'
../lib/libuhd.so.3.15.0: error: undefined reference to 'boost::serialization::typeid_system::extended_type_info_typeid_0::type_register(std::type_info const&)'

```

michaelld, 2019-05-13:
`/usr/bin/c++   -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG   CMakeFiles/uhd_image_loader.dir/uhd_image_loader.cpp.o  -o uhd_image_loader -Wl,-rpath,/home/omv/wspr/wsjtx/uhd/BUILD/uhd-3.14.0.0-20190513/host/build/lib: ../lib/libuhd.so.3.15.0 -lpthread -lrt -ldl /usr/lib64/libusb-1.0.so`

Notice: No Boost linkage. The Boost CMake scripts don't assign the expected return variables. Guessing they do assign some variables, just not sure which ones. Getting the return variables correct from any `FindPackage` call is critical to CMake scripts working correctly!

michaelld, 2019-05-13:
In my opinion (and not so humble even), if the Boost folks want to provide their own CMake find scripts, they should at least make them compatible with the CMake Boost find script, so that they are drop-in-compatible! The Boost CMake scripts could also assign other variables, of course, which the main CMake project scripts could choose to use, but this should be a -choice-. So either the Boost CMake scripts are not ready for prime time yet (because they aren't drop-in drop-in-compatible with the CMake Boost find script), or these scripts are intended to break the expected CMake `FindPackage` return variables. Either way it's a pain for us developers who want to use Boost 1.70.0 and CMake!

michaelld, 2019-05-13:
So the good news is:
(1) Boost 1.70.0 CMake find scripts are indeed broken. Boost has a fix already, which will be in the next release: https://github.com/boostorg/boost_install/commit/cfa8d55250dfc2635e907e42da423e4eb540dee5

(2) For UHD, we'll create a work-around for this specific version of Boost so that it works with all otherwise-compatible Boost versions including 1.70.0. I'll post a patch here for you to try once we have it working internally.

fedya, 2019-05-14:
Yes, fixed.

INFO for Maintainers:
if you want to build uhd
apply patch https://github.com/boostorg/boost_install/commit/cfa8d55250dfc2635e907e42da423e4eb540dee5
to boost and recompile it with -fPIC

michaelld, 2019-05-16:
Great. I'm closing this issue. We'll have an internal fix shortly.

closed 2019-05-16 by michaelld