EttusResearch/uhd #267: uhd-3.14.0.0 build failure with tests enabled

yarda, 2019-04-15:
## Issue Description ##

Build fails if tests are enabled, it builds OK with -DENABLE_TESTS=off.

## Setup Details ##

Fedora rawhide (f31), x86_64
gcc-9.0.1-0.12
boost-devel-1.69.0-6

## Expected Behavior ##

Build without failure

## Actual Behaviour ##

```
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `test_sph_send_one_channel_one_packet_mode::test_method()':
BUILDSTDERR: /builddir/build/BUILD/uhd-ac96d055091c29e2c6b08762b3b3d9da8db5b855/host/tests/sph_send_test.cpp:32: undefined reference to `mock_zero_copy::mock_zero_copy(uhd::transport::vrt::if_packet_info_t::link_type_t, unsigned long, unsigned long)'
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `test_sph_send_one_channel_full_buffer_mode::test_method()':
BUILDSTDERR: /builddir/build/BUILD/uhd-ac96d055091c29e2c6b08762b3b3d9da8db5b855/host/tests/sph_send_test.cpp:88: undefined reference to `mock_zero_copy::mock_zero_copy(uhd::transport::vrt::if_packet_info_t::link_type_t, unsigned long, unsigned long)'
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `mock_zero_copy::~mock_zero_copy()':
BUILDSTDERR: /builddir/build/BUILD/uhd-ac96d055091c29e2c6b08762b3b3d9da8db5b855/host/tests/common/../common/mock_zero_copy.hpp:73: undefined reference to `vtable for mock_zero_copy'
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `mock_zero_copy::~mock_zero_copy()':
BUILDSTDERR: /builddir/build/BUILD/uhd-ac96d055091c29e2c6b08762b3b3d9da8db5b855/host/tests/common/../common/mock_zero_copy.hpp:73: undefined reference to `vtable for mock_zero_copy'
BUILDSTDERR: collect2: error: ld returned 1 exit status
BUILDSTDERR: make[2]: *** [tests/CMakeFiles/sph_send_test.dir/build.make:100: tests/sph_send_test] Error 1
make[2]: Leaving directory '/builddir/build/BUILD/uhd-ac96d055091c29e2c6b08762b3b3d9da8db5b855/host/build'
BUILDSTDERR: make[1]: *** [CMakeFiles/Makefile2:2618: tests/CMakeFiles/sph_send_test.dir/all] Error 2
BUILDSTDERR: make[1]: *** Waiting for unfinished jobs....
```

## Steps to reproduce the problem ##

```
mkdir -p host/build
cd host/build
cmake ../
```

## Additional Information ##

michaelld, 2019-04-19:
Can you try the recently released Boost 1.70.0? I think you'll see the same build issues. Either way, can you record and attach the CMake configure output && verbose build log ("make VERBOSE=ON")? Having those would be really useful here.

mbr0wn, 2019-04-23:
Hey @yarda, thanks for the report. I know this is useful for making sure that UHD keeps building on Fedora. FYI, we do test UHD regularly on the latest Fedora, but usually not on rawhide.

I could not reproduce this issue. When I grabbed rawhide, I also got a different gcc version (patch level 0.14). However, all the tests built and ran. Is this still an issue?

yarda, 2019-04-23:
Sorry for delay, I was out during holidays :) I will retest and provide more info.

mbr0wn, 2019-04-23:
Thanks a lot! Appreciate your efforts.

yarda, 2019-04-23:
I am still unable to compile it with the tests enabled, the latest rawhide, the latest uhd git snapshot, full build log:
https://kojipkgs.fedoraproject.org//work/tasks/2248/34362248/build.log
Package versions used in the buildroot:
https://kojipkgs.fedoraproject.org//work/tasks/2248/34362248/root.log
There are no downstream patches.

Regarding the boost-1.70.0 I had a bit trouble to get it into to the rawhide build root, but I will try.

mbr0wn, 2019-04-23:
Is there an easy way for me to repro? I tried getting a rawhide Docker image, but there, it didn't fail. I think the Boost 1.70 is a bit of a red herring, unless the next Fedora will ship with it.

yarda, 2019-04-23:
The following reproducer should work from any Fedora:
```
# dnf install fedpkg mock
# fedpkg co --anonymous uhd
# cd uhd
# fedpkg srpm
# mock -r fedora-rawhide-x86_64 uhd-*.src.rpm
```
Then in the mock shell there is the build root for experiments:
```
# mock -r fedora-rawhide-x86_64 --no-clean --enable-network shell
```

yarda, 2019-08-03:
Same problem with uhd-3.14.1.0 on f31. It can be compiled without tests, i.e. the following works:
-DENABLE_TESTS=off 

With tests enabled it fails:
```
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `test_sph_send_one_channel_one_packet_mode::test_method()':
BUILDSTDERR: /builddir/build/BUILD/uhd-3.14.1.0/host/tests/sph_send_test.cpp:32: undefined reference to `mock_zero_copy::mock_zero_copy(uhd::transport::vrt::if_packet_info_t::link_type_t, unsigned long, unsigned long)'
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `test_sph_send_one_channel_full_buffer_mode::test_method()':
BUILDSTDERR: /builddir/build/BUILD/uhd-3.14.1.0/host/tests/sph_send_test.cpp:88: undefined reference to `mock_zero_copy::mock_zero_copy(uhd::transport::vrt::if_packet_info_t::link_type_t, unsigned long, unsigned long)'
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `mock_zero_copy::~mock_zero_copy()':
BUILDSTDERR: /builddir/build/BUILD/uhd-3.14.1.0/host/tests/common/../common/mock_zero_copy.hpp:73: undefined reference to `vtable for mock_zero_copy'
BUILDSTDERR: /usr/bin/ld: CMakeFiles/sph_send_test.dir/sph_send_test.cpp.o: in function `mock_zero_copy::~mock_zero_copy()':
BUILDSTDERR: /builddir/build/BUILD/uhd-3.14.1.0/host/tests/common/../common/mock_zero_copy.hpp:73: undefined reference to `vtable for mock_zero_copy'
BUILDSTDERR: collect2: error: ld returned 1 exit status
```

uhd-3.14.1.0
gcc-9.1.1
boost-1.69.0

Failed koji tasks with full build logs:
https://koji.fedoraproject.org/koji/taskinfo?taskID=36637924

mbr0wn, 2021-08-25:
I'm closing this -- 3.14 is just too old to do anything about it.

closed 2021-08-25 by mbr0wn