EttusResearch/uhd #270: Build UHD Python API 3.7.2 in Docker

tjhoeft, 2019-04-22:
## Issue Description ##
Unable to install UHD for Python 3.7. Works with Python 2.7

## Setup Details ##
Using: Docker version 18.09.5, build e8ff056
[python_uhd.dockerfile.txt](https://github.com/EttusResearch/uhd/files/3104546/python_uhd.dockerfile.txt)

bash> docker image build -f <dockerfile_name>

## Expected Behavior ##
Expect to be able to build UHD and on installation completion I would like to:

1. bash> python
2. python> import uhd

## Actual Behaviour ##
`
[100%] Linking CXX executable uhd_usrp_probe
[100%] Built target uhd_usrp_probe
[100%] Linking CXX executable usrp_burn_mb_eeprom
[100%] Linking CXX executable octoclock_burn_eeprom
[100%] Built target usrp_burn_mb_eeprom
[100%] Built target octoclock_burn_eeprom
[100%] Linking CXX executable responder
[100%] Built target responder
[100%] Linking CXX shared library libpyuhd.so
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(abstract.o): relocation R_X86_64_32S against `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.7m.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [python/libpyuhd.so] Error 1
python/CMakeFiles/pyuhd.dir/build.make:119: recipe for target 'python/libpyuhd.so' failed
CMakeFiles/Makefile2:4168: recipe for target 'python/CMakeFiles/pyuhd.dir/all' failed
make[1]: *** [python/CMakeFiles/pyuhd.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c make -j ${AVAILABLE_PROCESSORS} && make install && ldconfig' returned a non-zero code: 2
`

## Steps to reproduce the problem ##
Attempt to rebuild the Docker image

tjhoeft, 2019-04-22:
Found this interesting that if I symlink to Python3.7.2 and rebuild 
 - Using this: RUN cmake -DENABLE_PYTHON_API=ON ..
 - Instead of this: RUN cmake -DENABLE_PYTHON_API=ON -DENABLE_PYTHON3=ON -DPYTHON_EXECUTABLE:FILEPATH=/usr/local/bin/python3.7 ..
`
Things complete, but get shoved into the Python2.7 dist-packages
-- Installing: /usr/local/share/man/man1/uhd_usrp_probe.1.gz
-- Installing: /usr/local/share/man/man1/usrp_n2xx_simple_net_burner.1.gz
-- Installing: /usr/local/share/man/man1/usrp_x3xx_fpga_burner.1.gz
-- Installing: /usr/local/share/man/man1/usrp2_card_burner.1.gz
-- Installing: /usr/local/lib/python2.7/dist-packages/uhd
-- Installing: /usr/local/lib/python2.7/dist-packages/uhd/filters.py
-- Installing: /usr/local/lib/python2.7/dist-packages/uhd/types.py
-- Installing: /usr/local/lib/python2.7/dist-packages/uhd/libpyuhd.so
-- Installing: /usr/local/lib/python2.7/dist-packages/uhd/__init__.py
-- Installing: /usr/local/lib/python2.7/dist-packages/uhd/usrp.py
Removing intermediate container 4130c5ff34c4
`

mbr0wn, 2019-04-23:
@tjhoeft I'm not sure what you're saying with your second comment. And thanks so much for providing a Dockerfile, that makes reproducing very easy.
The first thing that jumps to my eyes is that you're trying a static build, that's not something we've ever tried or support (if this is very important, please contact us offline. I've kicked off a build with a shared library, but it'll take a while.

tjhoeft, 2019-04-23:
Thanks for the reply! Ok...I think a diagram is in order. See the attached
file. I'm still pretty new to building software from scratch. I'm not
concerned if the code is statically or dynamically linked. I only really
care if I can import UHD within a Python environment.

[image: Docker Build Scenarios.png]

-- 
*_____________________________________________________*

*TJ Hoeft*
Senior Research Engineer
p: 509-554-9933
a: 7021 Columbia Gateway Drive, Suite 175 Columbia, MD
<https://maps.google.com/?q=7021+Columbia+Gateway+Drive,+Suite+175+Columbia,+MD&entry=gmail&source=g>
 21046
<https://maps.google.com/?q=7021+Columbia+Gateway+Drive,+Suite+175+Columbia,+MD+%C2%A021046&entry=gmail&source=g>
w: department13.com  e: tjh@department13.com <spatel@department13.com>

CONFIDENTIALITY NOTICE This message and any included attachments are from
Department 13 and are intended only for the addressee. The information
contained in this message is confidential and any unauthorized forwarding,
printing, copying, distribution, or use of such information is strictly
prohibited and may be unlawful.

mbr0wn, 2019-04-23:
Looks like your image didn't integrate properly. However, by dynamic linking your Docker compiles (use `./configure --enable-shared`).

mbr0wn, 2019-04-23:
I'll close this for now, please reopen if that didn't help.

closed 2019-04-23 by mbr0wn